roleoroleo/yi-hack-Allwinner #176: Change wifi network

UlfgarIlFabbro, 2020-10-26:
Is it possible to change wifi network without resetting? Where is the setting located?

roleoroleo, 2020-10-26:
It's not possible.
The setting is saved inside a raw flash partition.

UlfgarIlFabbro, 2020-10-26:
to be able to reset and set the wifi again you have to put the original firmware or can I do it directly without removing yi-hack-Allwinner?

roleoroleo, 2020-10-26:
Directly without removing.

leocrawford, 2020-11-15:
The wifi creds are stored in /dev/mtdblock6 at byte 28 for SSID, and 92 for the key. You can see the current ones using:

    hexdump -C /dev/mtdblock6

and write them using:

    # write SSID
    printf $ssid | dd of=/dev/mtdblock6 bs=1 seek=28 count=64 conv=notrunc 
    # write key
    printf $key | dd of=/dev/mtdblock6 bs=1 seek=92 count=64 conv=notrunc

As both appear to be null terminated, if your new creds are shorter then the old ones, you will need to null pad, Most easily done by writing zeros to the space beforehand:

    cat /dev/zero  | dd of=/dev/mtdblock6 bs=1 seek=1 count=155 conv=notrunc

If easier, this script saved to `/tmp/configure_wifi.sh` (with `chmod +x` applied) should allow you to `/tmp/configure_wifi.sh <ssid> <key>`:

````
#!/bin/bash

ssid=$1
key=$2

function print_help {
	echo "configure_wifi.sh <ssid> <key>"
	echo "will be use don next boot"
}

if [ -z "$ssid" ]
then
	echo "\$error: ssid has not been set"
	print_help
	exit 1
fi

if [ -z "$key" ]
then
	echo "key: ssid has not been set"
	print_help
	exit 1
fi

# clear the existing passwords (to ensure we are null terminated)
cat /dev/zero  | dd of=/dev/mtdblock6 bs=1 seek=1 count=155 conv=notrunc
# write SSID
printf $ssid | dd of=/dev/mtdblock6 bs=1 seek=28 count=64 conv=notrunc
# write key
printf $key | dd of=/dev/mtdblock6 bs=1 seek=92 count=64 conv=notrunc
hexdump -C /dev/mtdblock6
````
I'd have offered a pull request, but I couldn't quickly see how to add a static resource to the build scripts.

roleoroleo, 2020-11-15:
Is there a field that stores if wifi procedure needs to be restarted?
Or setting user and password is enough?

leocrawford, 2020-11-15:
You don't need to do anything else, at least when you're changing the wifi password (i.e. it was already set up). I haven't tested it for a device that has never had wifi configured. If we can find such a device and dump `/dev/mtdblock6` we can see what else (if anything) changes.

Ideally, i guess, we'd look to in the SD card to see if there was a `wifi_settings` file, and if there was use it to run script. this would allow people to recover devices that had no/wrong wifi settings without reverting to original firmware.

roleoroleo, 2020-11-15:
I will study it.

roleoroleo, 2020-11-16:
If the ssid starts at 28 why do you erase from the beginning of the block?
This is my dump:

```
00000000  01 00 00 00 01 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 57 69 66 69  |............Wifi|
00000020  2d 53 53 49 44 00 00 00  00 00 00 00 00 00 00 00  |-SSID...........|
...
```

leocrawford, 2020-11-16:
No good reason other than it was already zeros, but you're right starting
at 28, for length 64+64 is the better answer.

roleoroleo, 2020-11-16:
I don't know if I will add a feature inside the hack (for example a web page to configure ssid and password).
The risk of bad writing on the flash is high.
Mostly in the encode/decode operations of special chars.

Probably it's better, as you said, to check and execute a script file in /tmp/sd

leocrawford, 2020-11-16:
If you do that, would it be useful to also allow mtd backup and restore on SD card?

If we had a `/tmp/sd/manage_yi_hack.conf` file with the following format:

````
mtd_backup=[1 2 3 6] # defaults to [6]
mtd_restore=[1 2] # defaults to []
wifi_ssid=
wifi_psk=
````

where mtd partitions could be backed up or restored to the sd_card by specifying their number, then it would allow people to revert bad wifi settings, assuming mtdblock6 was always backed up by default if `manage_yi_hack.conf` existed (which would be renamed after use). It would also make it easier for folk to share mtd partitions when filing issues, etc.

roleoroleo, 2020-11-21:
What do you think about?
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials

leocrawford, 2020-11-22:
Perfect. Nice one!

roleoroleo, 2020-11-22:
Did you try it?

leocrawford, 2020-11-23:
I didn't, but I read the scripts and it looked sound. I can do if you'd like an independent test?

The one thing i did wonder is if we need an auto-restore feature, e.g. boot cycle 1 write mtdblock6 and creates backup and renames or modified CFG file (to avoid writing on every boot by accident). If a user then renames the backup to restore.img or similar, boot cycle 2 could restore it.

roleoroleo, 2020-11-27:
I noticed that if I change the byte at offset 24 from 0 to 1 the cam requires the pairing.
So probably it's a "connection successfully" bit that must be changed.
I updated the script.

closed 2020-12-19 by roleoroleo

kikecalpe, 2022-10-30:
> What do you think about? [roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials)

This didn't worked for me.
To save time to anyone #289 did the trick.