roleoroleo/yi-hack-Allwinner #179: Problem with access to records via ftp

xrnv2, 2020-10-27:
Hello,

My settings:
disable cloud: true
Recording without cloud: true
Save video when a motion is detected: off

Data is not available via ftp but exist on sd card.

Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     10404      1372  88% /home
tmpfs                    32768        84     32684   0% /tmp
/dev/mmcblk0p1        31246752   5418656  25828096  17% /tmp/sd

root@yi-d8fc:/tmp/sd/record# ls
2020Y04M17D20H  2020Y10M26D21H  2020Y10M27D02H  2020Y10M27D07H  2020Y10M27D12H
2020Y04M17D22H  2020Y10M26D22H  2020Y10M27D03H  2020Y10M27D08H  2020Y10M27D13H
2020Y10M26D18H  2020Y10M26D23H  2020Y10M27D04H  2020Y10M27D09H  2020Y10M27D14H
2020Y10M26D19H  2020Y10M27D00H  2020Y10M27D05H  2020Y10M27D10H  2020Y10M27D15H
2020Y10M26D20H  2020Y10M27D01H  2020Y10M27D06H  2020Y10M27D11H  tmp.mp4.tmp

It is normal behavior? What I should do to share sd via ftp?

cchenga, 2020-10-27:
If you want the other ftp clients be able to access videos, you can enable FTP server in configurations -> Advanced

xrnv2, 2020-10-27:
Yes, ftp is enabled and seems to work fine, but share contains only empty 'topfile'. Probably this is /mnt and not contains any other files/folders

roleoroleo, 2020-10-28:
You can find the files in /tmp/sd/record

xrnv2, 2020-10-28:
Thank you guys for support. Path was available after setting password and using root user. I used anonymous user earlier and don't have access to dir /tmp/sd/record

closed 2020-10-28 by xrnv2