roleoroleo/yi-hack-Allwinner #180: FTP upload does not upload videos after motion detection

kimonberlin, 2020-10-30:
I'm seeing the same behavior with 0.2.0 and 0.2.1. I can see the videos using the web interface, but they never get uploaded.

Here's the ftp log on the server side:
Oct 30 12:58:07 homepi pure-ftpd: (?@192.168.1.151) [INFO] New connection from 192.168.1.151
Oct 30 12:58:07 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [user] [upload]
Oct 30 12:58:07 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [pass] [<*>]
Oct 30 12:58:07 homepi pure-ftpd: (?@192.168.1.151) [INFO] upload is now logged in
Oct 30 12:58:07 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [type] [I]
Oct 30 12:58:07 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [epsv] []
Oct 30 13:00:18 homepi pure-ftpd: (upload@192.168.1.151) [INFO] Logout.
Oct 30 13:00:18 homepi pure-ftpd: (?@192.168.1.151) [INFO] New connection from 192.168.1.151
Oct 30 13:00:18 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [user] [upload]
Oct 30 13:00:18 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [pass] [<*>]
Oct 30 13:00:18 homepi pure-ftpd: (?@192.168.1.151) [INFO] upload is now logged in
Oct 30 13:00:18 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [type] [I]
Oct 30 13:00:18 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [epsv] []
Oct 30 13:02:29 homepi pure-ftpd: (upload@192.168.1.151) [INFO] Logout.
Oct 30 13:02:29 homepi pure-ftpd: (?@192.168.1.151) [INFO] New connection from 192.168.1.151
Oct 30 13:02:29 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [user] [upload]
Oct 30 13:02:29 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [pass] [<*>]
Oct 30 13:02:29 homepi pure-ftpd: (?@192.168.1.151) [INFO] upload is now logged in
Oct 30 13:02:29 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [type] [I]
Oct 30 13:02:29 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [epsv] []
Oct 30 13:04:40 homepi pure-ftpd: (upload@192.168.1.151) [INFO] Logout.
Oct 30 13:04:40 homepi pure-ftpd: (?@192.168.1.151) [INFO] New connection from 192.168.1.151
Oct 30 13:04:40 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [user] [upload]
Oct 30 13:04:40 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [pass] [<*>]
Oct 30 13:04:40 homepi pure-ftpd: (?@192.168.1.151) [INFO] upload is now logged in
Oct 30 13:04:40 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [type] [I]
Oct 30 13:04:40 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [epsv] []

If I enable "Create directory tree", I see the directories get created properly, but videos are still not uploaded.

roleoroleo, 2020-10-30:
Try to run ftpput command manually.

kimonberlin, 2020-10-30:
> Try to run ftpput command manually.

root@yi-422f:/tmp/sd/record/2020Y10M30D13H# **ftpput --help**
-ash: ftpput: not found

I don't see a standalone ftpput, and "busybox ftpput" returns "ftpput: applet not found".

There's a wget, so if ftpput is available as a standalone file, I guess I could download it?

kimonberlin, 2020-10-30:
Ah, I found ftppush.sh and ftppush.log. Here's the log:

2020-10-30 [12-40-30] [INFO] === SERVICE START ===
2020-10-30 [12-40-31] [INFO] checkFiles
2020-10-30 [12-58-07] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y10M30D11H/24M06S54.mp4].
2020-10-30 [13-00-18] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y10M30D11H/25M00S60.mp4].
2020-10-30 [13-02-29] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y10M27D13H/31M52S08.mp4].
2020-10-30 [13-04-40] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y10M27D13H/32M00S60.mp4].
2020-10-30 [13-06-51] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y10M27D13H/33M00S60.mp4].
(...)

roleoroleo, 2020-10-31:
To run manually ftpput you have to source the environment:
```
. /home/yi-hack/script/env.sh
```

kimonberlin, 2020-10-31:
Thanks.
There's no server response after sending EPSV, and the connection eventually times out.

root@yi-422f:/home/yi-hack# ftpput --verbose --username upload --password ******** 192.168.1.7 /testfile.mp4 /tmp/sd/record/2020Y10M31D07H/55M00S60.mp4
Connecting to 192.168.1.7 (192.168.1.7:21)
ftpput: cmd (null) (null)
ftpput: cmd USER upload
ftpput: cmd PASS ********
ftpput: cmd TYPE I (null)
ftpput: cmd EPSV (null)
ftpput: can't connect to remote host (192.168.1.7): Operation timed out

On the server side:
Oct 31 09:50:12 homepi pure-ftpd: (?@192.168.1.151) [INFO] New connection from 192.168.1.151
Oct 31 09:50:12 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [user] [upload]
Oct 31 09:50:12 homepi pure-ftpd: (?@192.168.1.151) [DEBUG] Command [pass] [<*>]
Oct 31 09:50:12 homepi pure-ftpd: (?@192.168.1.151) [INFO] upload is now logged in
Oct 31 09:50:12 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [type] [I]
Oct 31 09:50:12 homepi pure-ftpd: (upload@192.168.1.151) [DEBUG] Command [epsv] []

kimonberlin, 2020-10-31:
Okay, I was just able to duplicate the issue with another pi as the ftp client. The ftp server stops responding after entering passive mode. This is not a yi-hack or busybox issue.

kimonberlin, 2020-10-31:
Solved on the server side by setting:
echo "30000 50000"  > /etc/pure-ftpd/conf/PassivePortRange
ufw allow 30000:50000/tcp

closed 2020-10-31 by kimonberlin