roleoroleo/yi-hack-Allwinner #181: YI Hack to MQTT with HomeAssistant intergration

EnricoFerro, 2020-10-31:
Hi,

I've created a small wrapper to transform the YI Hack All Winner webgui or ssh to a MQTT Provider.

The project is available to [EnricoFerro/yi-hack-to-mqtt](https://github.com/EnricoFerro/yi-hack-to-mqtt)

I use it to integrate the YI Hack camera with Home Assistant.

Hope this project can be useful for somebody.

Thanks,
Enrico

roleoroleo, 2020-10-31:
Great project!

EnricoFerro, 2021-03-05:
Hi,

I'm really bored about this project.. ..so I've created another project where I've reengineered the integration to work inside the camera.

The project is available to [EnricoFerro/yi-hack-to-mqtt-script](https://github.com/EnricoFerro/yi-hack-to-mqtt-script)

The reason for move the project is:
* There are a lot of information inside the camera that I've not have in a separate project
* The camera could be moved, renamed etc.. ...every time this information should be updated ( It is very boring, is better that the camera give the information )
* The old project occupy a lot of memory ( a lot of memory in a raspberry pi near the homeassistant ), the mew project use the ram of the camera ( and don't use so many ram )

In the future, I want to integrate this project inside the camera.. ..let me do some test.

Thanks,
Enrico

closed 2024-05-13 by github-actions[bot]