roleoroleo/yi-hack-Allwinner #182: 9FUS - ONVIF Blackscreen

Taranuka, 2020-10-31:
I've bought a Yi Camera Home 3 1080P and I flashed your firmware.

I've added the camera into the NVR/DVR GRT-K1104A. The IP-Camera is recognized and connected. I can change some options but the screen is black. It's working with VLC-Player and RTSP-Stream.

Did I missed something or did wrong?

roleoroleo, 2020-11-01:
Try to add the cam as rtsp (not onvif) and check if it works.

Taranuka, 2020-11-01:
I've tested the camera on my laptop with ONVIF Device Manager (ODM) and the RTSP with VLC. Both worked fine. I'm new to ip cameras. This Camera is really strange to me.

I have a MikroTik RouterBoard and Fritzbox. The IP-Camera is switching between two IP-Addresses (192.168.88.* and 192.168.178.*). My DVR have a 192.168.88.* IP. When I've connected it to my DVR/NVR with a 192.168.88.* and try to show me a picture with my camera restarts and got IP-adress with 192.168.178.*. Could this be the issue?

roleoroleo, 2020-11-01:
> I have a MikroTik RouterBoard and Fritzbox. The IP-Camera is switching between two IP-Addresses (192.168.88.* and 192.168.178._). My DVR have a 192.168.88._ IP. When I've connected it to my DVR/NVR [...]

I don't know but if there is a network problem, probably onvif does not respond at all. Not a blank screen

> Which username and password is used for SSH?

root without password if you didn't change it.

Taranuka, 2020-11-02:
Thank you. Both have now a IP-address with 192.168.88.* but it didnt solved my issue. The camera works fine with ONVIF Device Manager but shows only a blank screen if connected with the NVR. I did some research but didn't found anything than manuals for the GRT K1104A. ONVIF is backwards compatible?

NVR: ONVIF 2.2
Cam: ONVIF 2.0

roleoroleo, 2020-11-02:
Does your NVR support RTSP?

Taranuka, 2020-11-02:
I can access the analog cameras via RTSP looks like rtsp://NVR-Recorder:554/Streaming/Unicast/channels/401 but cant add a camera with rtsp-protocol. Only Grundig-1, Grundig-2 and ONVIF.

roleoroleo, 2020-11-03:
Unfortunately onvif is a complicated protocol. Each manufacturer implements a part of it and there are lot of incompatibilities.
If it works with odm I don't know how to help you.

roleoroleo, 2020-11-21:
Are you able to make a dump of the tcp traffic between the cam and your DVR?

Taranuka, 2020-11-26:
Do you know how I can do this? Could I make a dump with the Yi-Cam?

roleoroleo, 2020-11-26:
You need a tcpdump program.
Like wireshark in windows or tcpdump in linux.
This is the linux version for your cam:
[tcpdump.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5603703/tcpdump.gz)

closed 2021-02-15 by roleoroleo