roleoroleo/yi-hack-Allwinner #184: RTSP stream withou audio

roleoroleo, 2020-11-03:
@PieVo 
Could you please help me with this problem?

I ported your commit about audio on this platform.
I changed alsa-lib redirecting the output to a fifo and I changed RTSP server to read it.
The same work done on MStar.

It happens that sometimes audio doesn't start:

- When rmm starts it reads correctly the patched alsa-lib.
- rRTSPServer reads audio parameters and adds correctly to the audio subsession.
- When I play the stream from vlc I can see only the video track:
[image]

If I enable debug in the file WAVAudioFifoServerMediaSubsession.cpp
`#define DEBUG 1`
this is the log when it doesnt' work:
```
root@yi-hack-allwinner:/# RRTSP_AUDIO=yes RRTSP_RES=both rRTSPServer
"ch0_0.h264" stream, from memory
Audio enabled
Play this stream using the URL "rtsp://192.168.0.100/ch0_0.h264"

"ch0_1.h264" stream, from memory
Audio enabled
Play this stream using the URL "rtsp://192.168.0.100/ch0_1.h264"

"ch0_2.h264" stream, from memory
Audio enabled
Play this stream using the URL "rtsp://192.168.0.100/ch0_2.h264"
WAVAudioFifoSource found at x = 0
fReplicator->inputSource() = 0x703d0
Original source FMT: 33 bps: 219 freq: -1226645184
```

And this is the log when it works:
```
root@yi-hack-allwinner:/# RRTSP_AUDIO=yes RRTSP_RES=both rRTSPServer
fs: 8000
fs2: 8000

"ch0_0.h264" stream, from memory
Audio enabled
Play this stream using the URL "rtsp://192.168.1.243/ch0_0.h264"

"ch0_1.h264" stream, from memory
Audio enabled
Play this stream using the URL "rtsp://192.168.1.243/ch0_1.h264"

"ch0_2.h264" stream, from memory
Audio enabled
Play this stream using the URL "rtsp://192.168.1.243/ch0_2.h264"
WAVAudioFifoSource found at x = 1
fReplicator->inputSource() = 0xb6e1e140
Original source FMT: 1 bps: 16 freq: 8000
Create SimpleRTPSink: PCMU, freq: 8000, channels 1
```

In the first scenario bps and freq are wrong.

If I enable debug in the file DummySink.cpp it always works.
I don't know why but it always works.
If I'm not wrong debug only print a string:
```
#ifdef DEBUG
  if ((int)presentationTime.tv_sec % 10 == 0) {
    char uSecsStr[6+1]; // used to output the 'microseconds' part of the presentation time
    sprintf(uSecsStr, "%06u", (unsigned)presentationTime.tv_usec);
    envir() << "DummySink Presentation time: " << (int)presentationTime.tv_sec << "." << uSecsStr;
  }
#endif
```

PieVo, 2020-11-04:
That’s weird.. It’s a while ago, but because of the optional noise filter I
added a loop was added to iterate back over the filter chain. In the wrong
case: “WAVAudioFifoSource found at x = 0” it clearly failed and the result
is some garbage stream.. Perhaps the x=0 situation should result in a
failure.

If the debug only enables prints, then it is some timing related behavior.

Do you use the noise filter in the allwinner also? Otherwise you could
hardcode the correct wavesource.

Best regards P.

roleoroleo, 2020-11-04:
> If the debug only enables prints, then it is some timing related behavior.

I already tried to add a usleep but no success.
When I remove the print line:

`envir() << "DummySink Presentation time: " << (int)presentationTime.tv_sec << "." << uSecsStr;`

doesn't work.

> Do you use the noise filter in the allwinner also? Otherwise you could hardcode the correct wavesource.

No, I don't use noise filter. I will try...

roleoroleo, 2020-11-04:
Changing the gcc optimization option from -O2 to -O1 it works.

PieVo, 2020-11-04:
Interesting! So the loop that is selecting the proper filter is somehow
optimized away? Maybe add some pragma around this area to prevent the
compiler from bad interpretation...

closed 2020-12-19 by roleoroleo