roleoroleo/yi-hack-Allwinner #186: BFUS - Onvif black screen until view rtsp

casacsm, 2020-11-07:
My onvif dvr have black screen for camera, until I open 
at least one time vlc on pc and enter via rtsp.
After that onvif video was perfect!
Why? Can you help, please?

I cannot start - every time I reboot the camera - vlc on pc to activate onvif stream!!! ;)

My model is YI1080p home camera - YY.2016 - fw 8.2.0.* - hack y20ga ver. 0.2.1

roleoroleo, 2020-11-07:
Onvif is only a protocol descriptor.
When the cam provides the stream it uses rtsp.
So, if it works with vlc, the problem is probably the rtsp client on your dvr.

casacsm, 2020-11-07:
Thank you for your reply.

My dvr works with 4 different cameras using onvif protocol.
Now I want to add the yi camera.

I try to use rtsp on dvr to connect the yi camera, but I obtain black screen every time.

If I use onvif protocol on dvr, I obtain black screen. 
As soon as I connect to yi camera using rtsp on vlc, I see image on dvr.
I don't understand why.

roleoroleo, 2020-12-19:
What model is your dvr?

closed 2024-05-12 by github-actions[bot]