roleoroleo/yi-hack-Allwinner #187: Motion_detection_image topic suffix: no messages

Giulianini, 2020-11-07:
When I subscribe to the topic: "yicam-1/motion_detection_image" no message is sent.
If I subscribe to the topic:  "yicam-1/motion_detection" start and stop message are sent.

roleoroleo, 2020-11-07:
Are you using the last version?
Do you have human detection enabled?
I tested again now and it works.

Giulianini, 2020-11-07:
Yes I'm using the latest version. I tested it with human detection enabled and disabled and I listen to all topics looking for the message but no message containing the image is present: only motion_start/stop online are printed. I will investigate more and see if a message containing the image will arrive. Thanks for the quick response @roleoroleo

roleoroleo, 2020-11-07:
I'm using this command with mosquitto:
`mosquitto_sub -v -h 127.0.0.1 -t home-assistant/yi-cam/#`

Giulianini, 2020-11-08:
[image]

Logging with MQTT-FX is working. In my application I'm using Paho-Mqtt python library. I will investigate why it doesn't receive the message.

Giulianini, 2020-11-10:
I close the issue because now is working in Paho-mqtt.

closed 2020-11-10 by Giulianini