roleoroleo/yi-hack-Allwinner #188: FTP Delete (after upload) - NOT WORKING

cej0t4, 2020-11-10:
In version 0.2.0 the deletion of videos worked perfectly ... since 0.2.1 it does not delete the videos when uploading them to FTP.

roleoroleo, 2020-11-10:
Did you add the option in the web page?

cej0t4, 2020-11-10:
Yes, select the option in motion events. I have returned to 0.2.0 and everything works as before.

roleoroleo, 2020-11-10:
This is the only commit involving ftppush.sh:
https://github.com/roleoroleo/yi-hack-Allwinner/commit/282ec309f96a29e50b47be397cd1d508961e8ecb

cej0t4, 2020-11-11:
If you can try it with your hand with a camera ... in my two cameras the deletion of videos does not work after sending it by FTP

divadiow, 2020-11-12:
Same here :(

I've turned the feature on and off and rebooted several times.
I've yet to reset the cam to test

roleoroleo, 2020-11-12:
Fixed: https://github.com/roleoroleo/yi-hack-Allwinner/commit/7b590f02f02c635469f94b3a00789a803f9e0ecc
Try the new ftppush.sh

divadiow, 2020-11-22:
Thank you! I will try when I get the compiling prerequisites sorted.

isbkch, 2020-12-03:
Had the same issue. Tried the suggested commit and confirm it's working now

closed 2020-12-19 by roleoroleo