roleoroleo/yi-hack-Allwinner #189: nvr not displaying video

sander816, 2020-11-14:
i have installed the latest version on my yihome camera, everything is working, including rtsp video.
I can view the video in VLC.

Now i want to add it to my smar NVR, it detects its as onvif, but theres no image displayed.
What can i do?

cyraxnet, 2020-12-05:
I'm interested in hearing how you got on with this as I have noticed a similar issue.

I use rtsp to record to the NVR, but after some time the stream is lost in both live view and then of course recording too.  A reboot of the yi camera brings it back.

I have not had the time this week to work out if it is the NVR at fault, or the the rtsp stream.

UlfgarIlFabbro, 2020-12-15:
Maybe I understand the problem, the address used by the cam is rtsp: // IP ADRESS // ch0_0.h264, in my opinion it is wrong to put 2 // after the ip adress. I tried with android app and it sees me the video if I insert it / at the end of the ip. in my opinion just do not add the / at the end of the ip, is it possible to verify?

sander816, 2020-12-15:
Well, vlc works fine in full res
Also low res substream is working too

roleoroleo, 2020-12-15:
@UlfgarIlFabbro 
Where do you see double slash?

UlfgarIlFabbro, 2020-12-15:
On android with tinycam app it works and automatically double /

roleoroleo, 2020-12-15:
IMHO tinycam doesn't work very well with onvif.
Try onvifer.

cyraxnet, 2020-12-15:
> Well, vlc works fine in full res
> Also low res substream is working too

are you suggesting it is a problem with your nvr, if the stream is still available in vlc? i will confirm the same this weekend i.e. at the point the nvr can no longer access the rtsp stream, if it works in vlc, then there must be some implementation issue with the nrv and rtsp of the hack

closed 2024-05-12 by github-actions[bot]