roleoroleo/yi-hack-Allwinner #191: Flashing on Yi 1080p Home 9FUS	8.3.0 y25ga - did it work?

11001010111111101011101010111110, 2020-11-19:
After placing the latest release for Y25GA on an SD card and placing it in a camera then turning it on, the following end up on the SD card:

[image]

But from what I can tell, the camera still shows up in the Yi app and I can't reach it via HTTP, which I think indicates it didn't go well? What's the obvious thing that I am missing, @roleoroleo ?

Cheers and than you for your work, I look forward to sending you beer money.

roleoroleo, 2020-11-19:
Did you connect ro port 8080?

11001010111111101011101010111110, 2020-11-19:
I did. I am wondering if putting it on a guest network prevents me from accessing it locally.

With this firmware, should I still be able to access the camera via the Yi Home application. I put it on a guest network so Yi couldn't collect information on my normal network, can I safely put it on a non-guest network?

Does the image above tell you that the flash was successful? Firmware in Yi App shows as 8.3.0.0A_201912271707.

Thanks again for your help.

roleoroleo, 2020-11-19:
> With this firmware, should I still be able to access the camera via the Yi Home application. I put it on a guest network so Yi couldn't collect information on my normal network, can I safely put it on a non-guest network?

Yes, if you switch on "Disable cloud".

> Does the image above tell you that the flash was successful? Firmware in Yi App shows as 8.3.0.0A_201912271707.
Yes.

11001010111111101011101010111110, 2020-11-19:
Ah gotcha. I was able to make it work. The issue was on my end, having put my camera on the guest network. When you say "Disable Cloud" is that a setting for the camera specifically or in the Yi app in general?

closed 2020-11-20 by 11001010111111101011101010111110