roleoroleo/yi-hack-Allwinner #192: Feature request: zerotier support

nielscil, 2020-11-22:
Hi, would it be possible to add zerotier support for the cam(s)? I would like to use an external placed (other network and location) yi home camera in my synology surveillance station.

roleoroleo, 2020-11-22:
I don't know zerotier but if you want to try i compiled it.
[zerotier.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5579696/zerotier.tar.gz)

nielscil, 2020-11-23:
Thank you, I've tried the package you provided.  I've got the following error message when running:

` ERROR: unable to configure virtual network port: could not open TUN/TAP device: No such file or directory`

I think that tun isn't enabled/intergrated on the Yi home cam's, am I right? Would it be posssible to add tun?

roleoroleo, 2020-11-23:
If I'm not wrong, tun/tap device is a kernel feature and it's not available in yi cam.
Unfortunately we cannot build a new kernel.

nielscil, 2020-11-23:
Yeah, I believe it is a kernel module. It isn't possible to compile the kernel module and just enabling it?

roleoroleo, 2020-11-23:
I could try to build it but without a .config probably it doesn't work.

roleoroleo, 2020-12-21:
No luck, I build it but when I insmod the module the cam goes in kernel panic.

nielscil, 2021-01-12:
Alright, tnx for trying!

closed 2021-01-12 by nielscil