roleoroleo/yi-hack-Allwinner #193: New YI Home 1080P AI+ IFUSY44L9U with FW 9.0.19.02_202007141512 + MB Y29GA supported?

Max-Bel, 2020-11-24:
Is it possible to flash this camera with one of custom firmware?
@roleoroleo: The Allwinner-v2 could it works (https://github.com/roleoroleo/yi-hack-Allwinner-v2)?
Somebody has tested on this type of camera?

Thanks

roleoroleo, 2020-11-24:
allwinner-v2 should work.

Max-Bel, 2020-11-24:
I started with the backup of my camera frimware, by following the steps here: 
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)

After 5min on my sd card I didn't find the "/backup" folder, only "log" and "record" folders.

@roleoroleo, did something go wrong?

Max-Bel, 2020-11-24:
I repeated the procedure and it worked now.

Max-Bel, 2020-11-24:
The custom firmware y21ga_0.1.2 works fine also on my Y29GA mainboard. Thanks @roleoroleo

roleoroleo, 2020-11-24:
If the procedure works this modl is y21ga.
Don't rely on the number on the mainboard.

Max-Bel, 2020-11-24:
I'm still testing the firmware on my MB, some features don't work properly:
1-FTP Upload doesn't seem to work. I don't know how many time the camera uploads the file to FTP Server but in my case it never worked. 
2- The "AI Human Detection" never worked. I have enabled the MQTT and the value of the related topic always remains = "stop"

Any ideas?

Max-Bel, 2020-11-24:
root@YICamera_Sala:~# ftp
-ash: ftp: not found
root@YICamera_Sala:~# ftppush
-ash: ftppush: not found
root@YICamera_Sala:~# ptpput
-ash: ptpput: not found

@roleoroleo, Sorry, is this normal?

roleoroleo, 2020-11-24:
If you want to run it manually, you have to source env.
```
# . /home/yi-hack/script/env.sh
```

Max-Bel, 2020-12-20:
I'm testing 0.1.3 version of yi-hack-Allwinner-v2.
I activated MQTT, FTP upload and rtsp server. It works fine but each time the webSite not repond the all functionalities stop to work, I need restart. Any idea?

Thanks

roleoroleo, 2020-12-27:
Try to enable  swap file.

closed 2021-02-15 by roleoroleo