roleoroleo/yi-hack-Allwinner #194: Feature request: Nightvision toggle

okuzlu, 2020-11-24:
Is it possible to toggle the nightvision on a custom time to activate / disable it?

Reason it, that sensitivity is too high and it always changes from nightvision to normal and back so that it makes a annoying "click" sound.

roleoroleo, 2020-11-24:
You can only switch off ir leds.

EvgenPavlyuchek, 2020-12-17:
i thought if switch off ir leds then it must be in daynormal  but it continue change and do "click" sound
switch off ir leds dont work ?

roleoroleo, 2020-12-19:
Try using yi app and check if it works.

okuzlu, 2020-12-24:
I can confirm. I disabled ir led in yi app but it still make the "click" sound

tachyon142, 2021-03-13:
The click occurs when the camera engages or disengages the IR cut filter built into the lens housing. You can eliminate the clicking noise by electrically disabling this toggle. To do this, open the case, and carefully unplug the 2-pin connector which connects the lens housing to the pcb.

Note that the camera will still shift to black and white mode when it decides the environment is too dark for color, and may be less willing to switch back to color mode than before.

If a software enhancement is developed for this, individual control of the filter and B&W processing in ffmpeg would be ideal, but it is my understanding that changes from the default ffmpeg parameters isn't possible.

Regards

closed 2024-05-12 by github-actions[bot]