roleoroleo/yi-hack-Allwinner #197: Feature Request: Auto Reboot

cyraxnet, 2020-11-26:
It would be useful to add an auto-reboot option, such that you can set the camera to restart itself at a scheduled time.
This, to clear any memory leaks or connection issues with external devices, like an NVR.

I'm a part time OO-dev so if I can, I will try to implement it myself, but it might be a quick feature for someone already working with the project.

roleoroleo, 2020-12-19:
The last version contains crontab.
You can schedule reboot command.

closed 2020-12-19 by roleoroleo

cyraxnet, 2020-12-27:
For anyone that comes across this and needs the same, syntax is
`* * * * *  /sbin/reboot`

Obviously with the desired crontab schedule for preceding characters to /sbin/reboot