roleoroleo/yi-hack-Allwinner #198: Baby crying detection is not working

berzezo, 2020-11-27:
Hello,

I wanted to use baby crying detection with MQTT and Hassio.
Motion detection is working fine with the configuration.

Firmware Version | 0.2.1
Base Version | 8.2.0.0A_202007211802
Model Suffix | y20ga
Serial Number | BFUSY31GUV...
Local Time | Fri Nov 27 20:37:47 CET 2020
Uptime | 0h 19m 54s
Load Average | 1.51 1.17 0.78
Free/Total Memory | 26440/61164 KB
Free SD Space | 100%

Do you have any idea what can be the problem?

Thank you in advance.

WInetu007, 2020-12-03:
> Hello,
>
> I wanted to use baby crying detection with MQTT and Hassio.
> Motion detection is working fine with the configuration.
>
> Firmware Version | 0.2.1
> Base Version | 8.2.0.0A_202007211802
> Model [...]

Can You show me your configuration Motion detection in configuration.yaml and in camera motion events tab?
I try do this but this but binary_sensor.yihack_v4_motion_sensor is always off

berzezo, 2020-12-07:
Hello,

Hassio config:

binary_sensor:
    - platform: mqtt
      name: "Kamera mozgás Luca"
      state_topic: "babycam_luca/motion_detection"
      payload_on: "ON"
      payload_off: "OFF"
      availability:
       - topic: "babycam_luca/status"
         payload_available: "ONLINE"
         payload_not_available: "OFFLINE"
      qos: 1
      device_class: motion

[image]
[image]

berzezo, 2020-12-14:
Hi @roleoroleo ,

Do you have any idea what can be the problem with baby crying detection?
Thanks in advance.

roleoroleo, 2020-12-14:
No, I need to investigate.

roleoroleo, 2020-12-19:
Baby crying option disapperead from yi app.
Maybe yi removed this feature?

roleoroleo, 2020-12-19:
The last software changes the baby crying feature. Now it's a sound detection.
Ipc codes are changed also.
I will update the code.

berzezo, 2020-12-19:
Thanks for your work.

roleoroleo, 2020-12-21:
I updated the code.
Next release...

berzezo, 2021-02-07:
Sorry but, still not working.

closed 2021-03-27 by berzezo