roleoroleo/yi-hack-Allwinner #199: Install over older hack?

rnieuwen, 2020-12-01:
First of all, great job on this! I used this to flash a camera I thought I had bricked months ago, and it came back from the camera graveyard and is now performing brilliantly. That brings to my my question. I still have a few other cameras that have other Yi hacks running on them (I believe it is v3 but I am not sure). Is it possible to take your version and flash the cameras running an older yi hack version (orovided of course that the model is supported) or will that end in disaster? The reason I am asking is that your version performs way better than the older versions of the hack.

roleoroleo, 2020-12-01:
If it's an older version of another hack I think it's not compatible.
v3 and v4 work with hisilicon platform.
Mstar works with mstar platform.

rnieuwen, 2020-12-03:
Hi,

Thanks. I will check which platform my other caneras are running on to see if Your version would be compatible.

closed 2020-12-19 by roleoroleo