roleoroleo/yi-hack-Allwinner #203: Confirmed working: Yi Home 1080p Camera 3 (firmware 8.3.0.0A_201912271707) 9FUS Y32

tunerooster, 2020-12-04:
Just to let you know...
```
Yi Home 1080p Camera 3 (firmware 8.3.0.0A_201912271707) 9FUS Y32
```
* From: amazon.com/dp/B01CW4BVU4
* Installed: y25ga_0.2.2.tgz
* From: https://github.com/roleoroleo/yi-hack-Allwinner/releases

Not thoroughly tested yet, but it boots fine, all the services are running, and all the web pages display fine!  Great job roleoroleo!

tunerooster, 2020-12-05:
...having issues with rtsp now.  Intermittent connection failures.  Sometimes it fails to connect for minutes, then finally connects.  Even immediately after restarting rRTSPServer.  Is the latest rRTSPServer in the beta release?  Is there a newer version?  Please advise...

roleoroleo, 2020-12-05:
There are 2 little changes in the streaming code in the last version.
Please downgrade to 0.2.1 and check if it works correctly.

tunerooster, 2020-12-05:
There does not appear to be a:

* y25ga_0.2.1.tgz

So I don't think I can downgrade.  Please advise...

tunerooster, 2020-12-05:
If you can post a y25ga rRTSPServer executable for 0.2.1 I will try it.

roleoroleo, 2020-12-05:
Here it is:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5647364/rRTSPServer.gz)

tunerooster, 2020-12-06:
Thanks roleoroleo!

Provided rRTSPServer.gz installed and running...

I'll let you know if it works better...

Thanks again!

tunerooster, 2020-12-08:
I have been running a monitor on rRTSPServer 0.2.1 every 15 minutes for two days now and it has encountered *NO* problems (which is better even, than the *non-free* yi-hack-v4 version).  I installed it on the 0.2.2 release.  I am assuming what you posted is from the 0.2.1 release.  Anyway, it *now* works _great_!  **Thanks**!!!

Any chance it (the rtsp server) could be ported to the Hi3518e Chipset (i.e., Yi Outdoor)?  I can provide any dumps etc. you need...  For that matter, the whole release is better...  snapshots and many other features.

P.S. As you noticed, I failed to see the 0.2.1 release on the *releases* page, so thanks again for posting it separately.  :+1:

roleoroleo, 2020-12-08:
Yes, the version I posted is from 0.2.1. But I can't understand why the new version is worse.

The porting to Hi3518e is not simple, I don't know this platform and I don't know how the h264 driver works.
And finally, I don't own this cam.

tunerooster, 2020-12-08:
I never ran the 15 minute monitor on the 0.2.2 version.  The problems I encountered were intermittent, with it not responding based on manual tries.  Would you like me to re-activate rRTSPServer-0.2.2 and monitor it?  The camera is not yet deployed, so it would not be a problem...

Basically, the monitor calls ffmpeg to produce a 1 frame snapshot every 15 minutes.  If it fails, (i.e., if the snapshot file does not exist or is zero length) it reports an error.  Let me know if this would be useful...

roleoroleo, 2020-12-08:
Ok, try this new version.
It's aligned with the last commit.
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5660103/rRTSPServer.gz)

tunerooster, 2020-12-09:
Got it...  Installed and monitoring.  I'll let you know.

tunerooster, 2020-12-10:
I'm sorry to report that this version hardly works at all.  Over 24 hours, half the time it failed.  When I restart it, more often that not, it simply does not respond (connect). I.e., I hardly need a monitor to observe the problems.

Are there debug options you would like me to run?

roleoroleo, 2020-12-10:
I will check the differences...

roleoroleo, 2020-12-10:
Is audio stream available on both versions?
Is it possible thant 0.2.1 was running without audio?

Last try...
Please, try to run it manually with debug option.
It will produce a very long output but I could find something.

```
. /home/yi-hack/script/env.sh
RRTSP_DEBUG=1 RRTSP_RES=both RRTSP_AUDIO=yes rRTSPServer
```

tunerooster, 2020-12-11:
"Audio" and "Hires only" were set via the web interface so I assume they are being set for all tests.  I let wd-rtsp.sh start rRTSPServer...  I just killall to restart.

I will run with debug options and post the result.

roleoroleo, 2020-12-11:
Check with vlc (menu Tools -> Codec Information) if you have 2 tracks: video and audio.

tunerooster, 2020-12-11:
I don't have vlc installed, but ffmpeg (i.e. ffplay) have *NO* audio.  Here is the output from ffplay:

```
Input #0, rtsp, from 'rtsp://shed/ch0_0.h264':0KB sq=    0B f=0/0   
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 20 tbr, 90k tbn, 180k tbc
```
If there was audio, it would be listed as one of the streams.

**WAIT...**
 I just double checked and I was wrong.  I DO NOT HAVE AUDIO SET IN THE WEB INTERFACE.  So it is likely that 0.2.1 was running without audio, as was the latest 0.2.2.

I will turn in ON and recheck...

ffplay now displays:
```
Input #0, rtsp, from 'rtsp://shed/ch0_0.h264':0KB sq=    0B f=0/0   
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 20 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: pcm_mulaw, 8000 Hz, 1 channels, s16, 64 kb/s
```
Now *with audio*... just restarted and am getting no response.  Server is running and nmap says the port is open.  Looks like wd_rtsp.sh just killed it and it restarted.  Working again...

...still setting up to run DEBUG.  Where will the debug file be?

roleoroleo, 2020-12-11:
Debug is on stderr.
But it's enough, I wanted to make sure the test was in the same condition.
Or audio enable for both 0.2.1 and 0.2.2.
Or audio disable for both 0.2.1 and 0.2.2.

roleoroleo, 2020-12-11:
Changes from 0.2.1 to 0.2.2 about RTSP:

- Changed optimization flag from -O2 to -O1.
- Fixed a bug in cb_memcmp and cb_memmem functions.
- Changed memcpy function with custom cb2s_memcpy.
- Added algorithm to discard frames if the counter is lower than the previous one.

I could try to restore the previous code step by step if you are able to support me with a test (because my cam works correctly with 0.2.2).

tunerooster, 2020-12-11:
Let me re-confirm that 0.2.1 continues to work.  I.e., goes for 24 hours with no (or few) failures.  If so, I will run with DEBUG.  Assuming 0.2.1 continues to work, do you want me to run debug output on both 0.2.1 *and* 0.2.2?

Also, as the failures are intermittent, should I just keep trying to play the stream until it fails (i.e., while in debug mode)?  Sometimes it works for some amount of time... so that could produce very large files.

In any case, I would be happy to support you in any way you need, including step my step regression testing.

roleoroleo, 2020-12-11:
Yes, I need to compare the differences between 0.2.1 and 0.2.2.

Thank you.

roleoroleo, 2020-12-11:
Just a question, which client do you use to check the service?

tunerooster, 2020-12-12:
Ffmpeg

tunerooster, 2020-12-13:
I have been monitoring rRTSPServer-0.2.1 for over 24 hours now, without a single failure, just to confirm what I observer earlier.  I will run them both now as follows:
```
. /home/yi-hack/script/env.sh
RRTSP_DEBUG=1 RRTSP_RES=both RRTSP_AUDIO=yes rRTSPServer
```
The two executables (and their md5sum) I will monitored are:
```
0dae49ede090d85f1b776014ed3bdb9c  rRTSPServer-0.2.1
f1d17869a1d760c957dbfefccaa255ac  rRTSPServer-0.2.2-latest
```

roleoroleo, 2020-12-13:
Ok.
md5 is ok.

tunerooster, 2020-12-13:
As I feared, I am unable (so far) to get rRTSPServer-0.2.2-latest to fail while running the debug option per above.  Could it possibly be due to the options wd_rtsp.sh uses to run it?  I turned off RTSP via the web configuration interface before starting the tests, and AI am running it manually with the following command:
```
# . /home/yi-hack/script/env.sh; RRTSP_DEBUG=1 RRTSP_RES=both RRTSP_AUDIO=yes ./rRTSPServer-0.2.2-latest > /tmp/sd/log/rRTSPServer-0.2.2.debug 2>&1
```
But after several attempts using:
```
ffplay rtsp://shed/ch0_0.h264
and
ffmpeg -y -nostdin -loglevel quiet -i rtsp://shed/ch0_0.h264 -frames 1 "/tmp/snap.jpg"
```
from my desktop, it has not yet failed, and the log file is over 52MB...
I have attached the output for:
```
 rRTSPServer-0.2.1.debug
```
where I ran one ffplay and one ffmpeg snapshot.

[rRTSPServer-0.2.1.debug.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5684459/rRTSPServer-0.2.1.debug.gz)

I will keep trying...

tunerooster, 2020-12-13:
I doubt I have the disk space on the sd card to run the test all day.  I am wondering if there are some strings in the debug output I could filter the output through before writing it to disk.  I.e.,
```
rRTSPServer-0.2.2-latest 2>&1 | grep -e pattern-1 -e pattern-2 -e etc... >  ./rRTSPServer-0.2.2-latest.debug
```
This could make it feasible to run all day until it fails.

tunerooster, 2020-12-13:
P.S.  Or possibly grep -v to exclude all the repetitive stuff...

tunerooster, 2020-12-13:
Wait!!!  I just got a failure.  ffplay and ffmpeg are both hung waiting for a reply.  I have included the output...

[rRTSPServer-0.2.2.debug.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5684475/rRTSPServer-0.2.2.debug.gz)

roleoroleo, 2020-12-13:
Thanks to your log, I found a bug.
I don't know if it fixes your problem but...
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5684500/rRTSPServer.gz)

tunerooster, 2020-12-13:
Great (hopefully)!  I will install and monitor all day...

tunerooster, 2020-12-14:
...have been monitoring 24 hours or so without a single failure!  It seems you have nailed it.  Great work!

tunerooster, 2020-12-14:
P.S. Would this bug/fix apply to the MStar release as well?  I have a camera running Mstar... (but have not yet tested it).

roleoroleo, 2020-12-14:
> Out of curiosity, what was the bug?

When I added the control about frame counter I forgot a line.
My fault.

> P.S. Would this bug/fix apply to the MStar release as well? I have a camera running Mstar... (but have not yet tested it).

This patch is not applicable, MStar has a completely different approach.

closed 2020-12-14 by tunerooster