roleoroleo/yi-hack-Allwinner #204: yi-hack-Allwinner 0.2.2. can't access the rtsp server which seems to be inaccessible

LIR131, 2020-12-05:
Hi everybody,
I need dome help.
i use the new yi-hack-Allwinner 0.2.2.
After about 3h of running the client can't access the rtsp server which seems to be inaccessible, i need to restart ffmpeg could you help me please?

That's the message

[NULL @ 0x750560] missing picture in access unit with size 27bitrate=N/A dup=361 drop=182 speed=   1x
[h264 @ 0x7c25d0] no frame!
rtsp://192.168.1.11/ch0_1.h264: Connection timed out
rtsp://192.168.1.11/ch0_1.h264: Connection timed out44:47.44 bitrate=N/A dup=361 drop=182 speed=0.999x
rtsp://192.168.1.11/ch0_1.h264: Connection timed out44:47.44 bitrate=N/A dup=361 drop=182 speed=0.998x
rtsp://192.168.1.11/ch0_1.h264: Connection timed out44:47.44 bitrate=N/A dup=361 drop=182 speed=0.997x
Error while decoding stream #0:0: Invalid data found when processing input
rtsp://192.168.1.11/ch0_1.h264: Connection timed out44:47.44 bitrate=N/A dup=361 drop=182 speed=0.996x
frame=197751 fps= 20 q=-1.0 Lq=-1.0 size=N/A time=02:44:47.56 bitrate=N/A dup=361 drop=182 speed=0.994x
video:352141kB audio:64424kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown

jjansen85, 2020-12-05:
Also an problem with the RTSP stream. Sometimes it stops streaming for about 30 minutes. The past 12 hours it stopped about 4 times.
Problem started after update to 0.2.2

roleoroleo, 2020-12-05:
@LIR131 
With 0.2.1 did you noticed the same problem?

Are you able to start rRTSPServer in debug mode and send me the output?
`RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 RTSPServer`

LIR131, 2020-12-05:
[image]
Do you mean those informations? if not, can you explain to me how can i do?

LIR131, 2020-12-05:
> 
> 
> @LIR131
> With 0.2.1 did you noticed the same problem?
> 
> Are you able to start rRTSPServer in debug mode and send me the output?
> `RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 RTSPServer`

LIR131, 2020-12-05:
i did not notice that in the 0.2.1 but i notice that the 0.2.2 loose less packet than the 0.2.1

tunerooster, 2020-12-05:
I am also having issues with RTSP.

* y25ga_0.2.2.tgz

It does not appear that 0.2.1 is available for y25ga.  From my 203 posting: 

Getting intermittent connection failures. Sometimes it connects immediately. Sometimes it fails to connect for minutes, then finally connects. Even immediately after restarting rRTSPServer.

roleoroleo, 2020-12-05:
0.2.1 is available in the releases section.

tunerooster, 2020-12-08:
Ah, I see it now.  I did not scroll down...  Only saw "Assets" section.

LIR131, 2020-12-18:
> Ah, I see it now.  I did not scroll down...  Only saw "Assets" section.
> 
@roleoroleo 
Hi,
I give you a feedback tomorrow.

cyraxnet, 2020-12-22:
> @LIR131
> With 0.2.1 did you noticed the same problem?
> 
> Are you able to start rRTSPServer in debug mode and send me the output?
> `RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 RTSPServer`

I am having the same issue, on 0.2.1. I have updated one of my cameras to 0.2.2 to see if the problem goes away.  I will report back tomorrow.  I mentioned the issue in #189, but basically I think it boils down to rtsp stream randomly becoming unavailable. Could be after a few mins, or few hours but eventually stream is lost.  Restart of the rtsp service fixes the issue.

I am brand new to this project. I would like to grab the log @roleoroleo.  Im having trouble starting the service with debug.  I can SSH to the camera. Do I need to cd into a specific directory to launch rRTSPServer?  I also assume you have a typo and it should be
`RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 rRTSPServer`

roleoroleo, 2020-12-23:
You need to source the env:
`. /home/yi-hack/script/env.sh`
The run the command:
`RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 rRTSPServer`

cyraxnet, 2020-12-23:
Ah yes, that worked sourcing the env.

Can I output the log to to a file on the sd card? When I run the debug from ssh, it prints to terminal, so might miss something. Anyways, I didn't catch anything. I just see scene information.

```
1608715921934: frame_len: 725 - frame_counter: 7830 - resolution: 360
1608715921934: frame detected - frame_len: 1683 - frame_counter: 7827 - frame_counter_last_valid: 7827 - resolution: 1080
1608715921959: frame_len: 1683 - cb_current->size: 262144
```
However, it definitely stopped streaming in the hour I went away and came back.  The stream stopped being available form my NVR. I checked then using vlc (with logging):

```
live555 error: Failed to connect with rtsp://x.x.x.x:554/ch0_0.h264
main error: connection failed: Connection refused by peer
satip error: Failed to connect to RTSP server x.x.x.x:554
main error: connection failed: Connection refused by peer
access_realrtsp error: cannot connect to x.x.x.x:554
```
This is with y20 on 0.2.2

roleoroleo, 2020-12-23:
You can redirect the output to a file or save the log with putty.

cyraxnet, 2020-12-24:
Same again. This time I caught it on y20 0.2.1 which I have left for testing against 0.2.2.

I saved putty output to disk, and although there are no timestamps (I know when the nvr reported the stream stopped working), reviewing the log there is no other info than what I reported above about stream information. No errors or warns etc.

Reviewing the logs from the NVR, annoyingly there is nothing for the period in which the recordings stopped.  I am going to try to find a simple rtsp viewer and let that run, see if I can get more logging at the point stream stops working.

cyraxnet, 2020-12-24:
I should perhaps open my own issue for tracking, but after further investigation, and happening to spend time on the laptop today, I think there are two issues, in my case at least.

1. Possible RTSP server issue - unknown at this time but the NRV confirms the device is online but no stream is available
2. The camera is dropping network connectivity/stuck in reboot loop - I happen to have been running a ping -t and noticed the one camera would power on, "uptime" would be around 1minute, then the device would lose connection (no pings, http became unavailable) and then it would reboot.  This is confirmed with new "uptime" value, and cycle repeats but would never come back online properly, until I did a manual reboot of the camera myself. Then the device came back in a healthy state.

Note, this camera is very close to the router so I do not think it is a signal issue. RAM was reporting the below during the bad cycles. 
```
"total_memory":"61164",
"free_memory":"27564",
```
I wasn't able to confirm "load_avg".  I will enable swap file on this camera, and see if this stops random reboots.

It would be worth others having similar issues to work out which of the above two they are seeing.

roleoroleo, 2020-12-27:
Maybe a memory leak?
Open another issue for this.

LIR131, 2020-12-27:
> 
> 
> You need to source the env:
> `. /home/yi-hack/script/env.sh`
> The run the command:
> `RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 rRTSPServer`

@roleoroleo 
How can I source the env?
Excuse for the delay

roleoroleo, 2020-12-28:
Open a ssh terminal with putty and login.
Set the logging feature of putty so that you can save the ouput of the screen.

Source the env with this command:
`. /home/yi-hack/script/env.sh`
Kill the running processes:
```
killall wd_rtsp.sh
killall rRTSPServer
```
Run the server in debug mode:
`RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 rRTSPServer`
Wait for the problem, stop logging and check the log file.

LIR131, 2020-12-30:
@roleoroleo 
I got it after 20minutes,but it's a huge file. 26Mo i Zip it
[puttycamsalon.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/5756015/puttycamsalon.zip)
[image]

roleoroleo, 2020-12-31:
Maybe I found the problem.
Try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5757089/rRTSPServer.gz)

LIR131, 2020-12-31:
Where should I put it and how?

Cordialement E.VEREPLA
Best regards

roleoroleo, 2020-12-31:
Connect with ssh and kill the processes:
```
killall wd_rtsp.sh
killall rRTSPServer
```
Rename rRTSPServer:
```
mv /home/yi-hack/bin/rRTSPServer /home/yi-hack/bin/rRTSPServer.ori
```
Copy with ftp the new rRTSPServer in the folder /home/yi-hack/bin/
Change the file permissions:
```
chmod 0755 /home/yi-hack/bin/rRTSPServer
```

LIR131, 2020-12-31:
@roleoroleo 
When i upload the new file and try to restart in debug i got :

root@yi-6337:~# chmod 0755 /home/yi-hack/bin/rRTSPServer
root@yi-6337:~# RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 rRTSPServer
-ash: rRTSPServer: not found
Do I have forgot somethings?

LIR131, 2021-01-01:
yesterday I installed the binary.
After 17 hours It's still running.
First impression : the speed is constant.
Scond: it's seems that the issues diseappear. 
But more than 15 times in 17 hours i got somethings like:
[image]
I have to stop it now to enhance somethings on my streaming server and then l restart it for long time.
I tell you more on saturday.

LIR131, 2021-01-02:
@roleoroleo 
Unfortunately after restarting the rstp server stuck after 3hours, 
It seems to be a very random issue.
I send you the server log:
[puttycamsalon.log.gz.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5760689/puttycamsalon.log.gz.gz)
and my streaming log when loosing connection to the rstp server:
[image]

roleoroleo, 2021-01-03:
I fixed another bug but I don't know if this solve your problem.
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5761725/rRTSPServer.gz)

LIR131, 2021-01-04:
@roleoroleo 

It still running after 5h and the connexion is still alive.
I'll tell you more after 1day running.

michalbogacz, 2021-01-07:
I have similar issue with rtsp and rtsp server from this thread is much more stable than last release (I'm testing it from yesterday).
@roleoroleo are you planing to release fix soon?

roleoroleo, 2021-01-08:
I will try to release the next weekend.

LIR131, 2021-01-09:
@roleoroleo 

I confirm, until now (after several hours) i did not get the connection timeout error

Your fix seems to be good

nmenoux, 2021-01-13:
@roleoroleo I confirm your last rRTSPServer.gz seems working as I do not have any more issues.

However, I have another YI Home camera with a new firmware (9.0.19.02_202007141512) , it is working with all-Winner-2 but the rRTSPServer has the same issue and your update seems not working for this release. Do you have source changes for recompiling it for this new firmware ? 

Thanks for your great work @roleoroleo ! My connected chicken coop (https://www.twitch.tv/cloudaryer) is working like a charm with your Yi hack :-)

roleoroleo, 2021-01-14:
> However, I have another YI Home camera with a new firmware (9.0.19.02_202007141512) , it is working with all-Winner-2 but the rRTSPServer has the same issue and your update seems not working for this release. Do you have source changes for recompiling it for this new firmware ?

Could you open an issue in the AllWinner repo?
I will check it.

cyraxnet, 2021-04-03:
I have been away from this project, after becoming busy with life, but I recently upgraded the cameras to latest version and this is resolved so thank you very much.
The small number of dropouts I get now I can believe are related to poor wifi or transient issue. No longer are there hours of stream recording missing. Great work!

closed 2021-05-15 by roleoroleo