roleoroleo/yi-hack-Allwinner #205: After upgrade 0.2.2 got 404 not found

rt400, 2020-12-05:
Hi,
i just did firmware upgrade from camera web ui and after restart i got error 404 not found..
any idea ??

m4rtyn, 2020-12-07:
Hello,
I can confirm that my unit also shows a 404 error at the web ui after upgrading from 0.2.1 to 0.2.2 (Y20).
The RTSP stream seems to work fine.

roleoroleo, 2020-12-08:
My y20 works correctly.
Please login with ssh and check if httpd daemon is running.

rt400, 2020-12-08:
> My y20 works correctly.
> Please login with ssh and check if httpd daemon is running.

Sorry but I'm noob in Linux, but would great if you could tell me what command to write?

m4rtyn, 2020-12-08:
I've disabled ssh due to high load on the camera. It seems like the httpd daemon is running because I get a login prompt and a 404 page, no timeout.

rt400, 2020-12-08:
> I've disabled ssh due to high load on the camera. It seems like the httpd daemon is running because I get a login prompt and a 404 page, no timeout.

same to me but i didn't disable the ssh

roleoroleo, 2020-12-08:
Please, post your config.
I will try to load it to my cam.

pinguinpfleger, 2020-12-08:
same here. cam kept offline after update.
```
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=yes
NTPD=yes
NTP_SERVER=192.168.0.1
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=kayi
PASSWORD=yikipw10
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=3
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
SSH_PASSWORD=erg,,,t4M-.JU7-HAHAHRRRRR-6lgef
```

pinguinpfleger, 2020-12-08:
here the same, cams are offline after update, no ping
```
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=yes
NTPD=yes
NTP_SERVER=192.168.0.1
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=PISQETHGWRTEHSER
PASSWORD=ehwrthjtrhn0í
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=3
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
SSH_PASSWORD=erg,,,t4M-.JU7-HAHAHRRRRR-6lgef

```

rt400, 2020-12-10:
@roleoroleo any chance to fix it ?

roleoroleo, 2020-12-10:
The config is ok.
Login with ssh or telnet and check:
- if httpd is running
- the content of the folder /home/yi-hack/

```
ps
ls -la /home/yi-hack
```

rt400, 2020-12-10:
ps : 
`root@DoorCamera:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  179 root         0 SW   [oom_reaper]
  180 root         0 SW<  [writeback]
  182 root         0 SW   [kcompactd0]
  183 root         0 SW<  [crypto]
  184 root         0 SW<  [bioset]
  186 root         0 SW<  [kblockd]
  224 root         0 SW   [sys_user]
  232 root         0 SW<  [cfg80211]
  239 root         0 SW<  [watchdogd]
  253 root         0 SW<  [spi0]
  263 root         0 SW   [kswapd0]
  348 root         0 SW<  [bioset]
  353 root         0 SW<  [bioset]
  358 root         0 SW<  [bioset]
  363 root         0 SW<  [bioset]
  368 root         0 SW<  [bioset]
  373 root         0 SW<  [bioset]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  402 root         0 SW   [irq/302-sunxi-m]
  404 root         0 SW   [irq/166-sdc0 cd]
  405 root         0 SW   [irq/303-sunxi-m]
  463 root         0 SW<  [kworker/0:1H]
  472 root         0 SW<  [bioset]
  473 root         0 SW   [mmcqd/0]
  478 root         0 SWN  [jffs2_gcd_mtd3]
  483 root       980 S    -/bin/sh
  526 root         0 SW   [ksdioirqd/mmc1]
  529 root         0 SW   [RTW_XMIT_THREAD]
  530 root         0 SW   [RTW_CMD_THREAD]
  531 root         0 SW   [RTWHALXT]
  532 root       680 S    ./log_server
  533 root       984 S    ./dispatch
 1033 root     39944 S    ./rmm
 1406 root         0 SW   [kworker/u2:1]
 1533 root      3004 S    ./mp4record
 1545 root       844 S    httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.co
 1563 root       844 S    pure-ftpd (SERVER)
 1575 root       876 S    dropbear -R -B
 1579 root       668 S    ipc_multiplexer
 1591 root       892 S    mqttv4
 1621 root       844 S    ntpd -p 192.168.1.254
 1634 root      5940 S    rRTSPServer
 1663 root       984 S    {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 1704 root      8044 S    onvif_srvd --pid_file /var/run/onvif_srvd.pid --mode
 1715 root      2080 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --
 1716 root      2420 S    framefinder
 1725 root       848 S    /home/yi-hack/usr/sbin/crond -c /var/spool/cron/cron
 4458 root       900 R    dropbear -R -B
 4487 root       980 S    sleep 10
 4506 root       980 S    -ash
 4535 root       980 R    ps
 7109 root         0 SW   [kworker/0:1]
 7139 root      1400 S    /home/base/tools/wpa_supplicant -c/tmp/wpa_supplican
 7475 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.scrip
21149 root         0 SW   [kworker/u2:0]
`
ls -la /home/yi-hack : 
`root@DoorCamera:~# ls -la /home/yi-hack
drwxr-xr-x    9 root     root             0 Dec  5 16:03 .
drwxr-xr-x    7 root     root             0 Jan  1  1970 ..
drwxr-xr-x    2 root     root             0 Dec  5 16:03 bin
drwxr-xr-x    4 root     root             0 Jan  1  1970 etc
drwxr-xr-x    2 root     root             0 Dec  5 16:03 lib
-rw-r--r--    1 root     root             6 Dec  5 16:03 model_suffix
drwxr-xr-x    2 root     root             0 Dec  5 16:03 sbin
drwxr-xr-x    2 root     root             0 Dec  5 16:03 script
drwxr-xr-x    4 root     root             0 Dec  5 16:03 usr
drwxr-xr-x    5 root     root             0 Dec  5 16:03 www
`

roleoroleo, 2020-12-10:
httpd is running and www is present.
Try
`ls -la /home/yi-hack/www`

rt400, 2020-12-10:
> httpd is running and www is present.
> Try
> `ls -la /home/yi-hack/www`

root@DoorCamera:~# ls -la /home/yi-hack/www
drwxr-xr-x    5 root     root             0 Dec  5 16:03 .
drwxr-xr-x    9 root     root             0 Dec  5 16:03 ..
drwxr-xr-x    2 root     root             0 Dec  5 16:03 css
drwxr-xr-x    2 root     root             0 Dec  5 16:07 js
drwxr-xr-x    2 root     root             0 Dec  5 16:03 pages
lrwxrwxrwx    1 root     root            14 Dec  5 16:03 record -> /tmp/sd/record

roleoroleo, 2020-12-10:
Your www dir is not ok.
cgi-bin is missing, index.html is missing.
etc...

Apply again the hack, something went wrong during update.

rt400, 2020-12-12:
i can confirm after i add the hack manually its work again.
i close this issue..

closed 2020-12-12 by rt400