roleoroleo/yi-hack-Allwinner #206: y25ga (releases 0.2.1 and 0.2.2) Snapshot script crashes camera

74ls04, 2020-12-06:
[image]

**Replicated on Release 0.2.1 and 0.2.2**

It seems that the snapshot script crashes the camera. I first observed this when setting up MQTT on Release 0.2.2 -- I noticed it crashed right after sending a motion event. I checked the configuration and noticed that one of the MQTT fields sends a raw jpeg image, which isn't present on my y20ga cameras, and that pointed me to the true issue. I flashed Release 0.2.1 and observed the same issue. 

Steps to replicate (method 1):
- Visit the snapshot url http://#.#.#.#:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
- Refresh the browser to call script again
- The camera crashes

Steps to replicate (method 2):
- Configure MQTT motion event
- Trigger motion event
- The event gets sent but camera crashes

I looked in the sd/logs directory but didn't see anything related -- not sure if that's where to look for logs.

74ls04, 2020-12-06:
I was able to get it to crash by calling the imggrabber program directly. In the screenshot below I called it three times in a row and it crashed after the third call. 

[image]

roleoroleo, 2020-12-06:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/5

74ls04, 2020-12-06:
Thanks, not sure how I missed that! Enabling swap does stop the crashing. Looks like this newer camera firmware uses up a few more resources.

closed 2020-12-06 by 74ls04