roleoroleo/yi-hack-Allwinner #207: yi-home 1080 outdoor future support?

Giulianini, 2020-12-06:
You plan to support the outdoor version of yi-home? The-Crypt0 is doing nothing in his project and I regrets donating my money to him. I prefer donating money to people that deserve it.

roleoroleo, 2020-12-06:
It depends on the platform.
If it's based on hisilicon platform, no.

Sprocket02, 2020-12-09:
I also have two outdoor cameras of that type. I think these cameras are based on the processor, but how do I find out if it's real?

roleoroleo, 2020-12-09:
Open it and take a picture of cpu.
Or make a dump of the flash.

Sprocket02, 2020-12-09:
Sorry for the stupid question but how do I dump the flash?

roleoroleo, 2020-12-09:
It depends on the platform.
This is the procedure for MStar: https://github.com/roleoroleo/yi-hack-MStar/wiki/Dump-your-backup-firmware

But, 1st step, take a picture.

Sprocket02, 2020-12-09:
Oh, thanks for the description, but this is a bit too much for me. Are there different types of processors in the YI Outdoor? I am using the yi-hack-v4 today, does that perhaps help?

Giulianini, 2020-12-09:
I will try to take a picture of the chip in a near future

Giulianini, 2020-12-09:
I forgot to mention that I'm using yi-hack-v4 like @Sprocket02. @roleoroleo this would help?

roleoroleo, 2020-12-09:
> 
> 
> Oh, thanks for the description, but this is a bit too much for me. Are there different types of processors in the YI Outdoor? I am using the yi-hack-v4 today, does that perhaps help?

If you are using yi-hack-v4, your cam is hisilicon based.
This hack doesn't support this plaform.

closed 2020-12-27 by roleoroleo