roleoroleo/yi-hack-Allwinner #210: 9CUSY Firmware available?

lordmathew, 2020-12-08:
Hello,
My camera has the serial started with 9CUSY. 
I've try all the firmwares and none of them works.
Is there any alternative or do i need to wait for a new firmare release?

TIA,
Nuno

roleoroleo, 2020-12-08:
The serial is not enough to know if there is an hack.
Firmware version?
Home, dome, ,...?

lordmathew, 2020-12-08:
Sorry.
Outdoor 1080p
Model: YHS.3017
Serial: 9CUSY36MGW
FW: 3.0.0D_201809111054
Thx

roleoroleo, 2020-12-08:
Your cam is not supported from my hack: MStar, Allwinner and Allwinner-v2.
And I don't know what platform it's based on.

lordmathew, 2020-12-09:
How can i help to identify the platform based on?
Pictures from the camera:  
https://prnt.sc/vyw74w
https://prnt.sc/vyw9mb

roleoroleo, 2020-12-09:
I need a picture of the internal board.

closed 2021-02-15 by roleoroleo