roleoroleo/yi-hack-Allwinner #213: Confirmed working: Yi Home 1080p 8.2.0.0A_201912270941 y20ga BFCNY31

BerlinChan, 2020-12-12:
## Device Info

```
Yi Home 1080p
Firmware Version | 0.2.2
Base Version | 8.2.0.0A_201912270941
Model Suffix | y20ga
Serial Number | BFCNY31...
```

---

I tested these functions are all working, maybe this model can added to Supported Cameras section on README file.

- RTSP High Res with audio
- snapshot
- MQTT
- Disable Cloud
- Record without Cloud

Thank you for your working.

closed 2020-12-27 by roleoroleo

2k13yr, 2021-02-16:
@BerlinChan Could you please share link for purchase the camera device of YI Home 1080p BFCNY31 ? I don't know which specified device is BFCNY31 on shopping websites. Thanks!

BerlinChan, 2021-02-17:
> @BerlinChan Could you please share link for purchase the camera device of YI Home 1080p BFCNY31 ? I don't know which specified device is BFCNY31 on shopping websites. Thanks!

I have buy this YI Home 1080p (Serial Number: `BFCNY31`) at **13, Sept, 2020** on Taobao.com.  I checked purchase history, on webpage snapshot it shows model **Y20**, and there is no description about device's `Serial Number` and `Base Version` info.

Now, only model **Y4** at selling on the product page, I have also buy. It's Serial Number is start with `BFCNY39`, and Base Version `9.0.09.03_202005111854`.

I installed firmware [Yi hack Allwinner v2](https://github.com/roleoroleo/yi-hack-Allwinner-v2) (firmware version 0.1.3) , I can access web UI, but RTSP server and snapshot doesn't working.