roleoroleo/yi-hack-Allwinner #215: After upgrading to 0.2.2 camera stopped recroding motion events

Adambl4, 2020-12-16:
I've upgdared my camera from 0.2.1 to 0.2.2 from web interface and after that it stopped saving any videos to `/tmp/record` folder. 
I've reset it to default several times and have reinstalled firmware manually by copying firmware files to sd card root.
In `ipc_multiplexer` output I see a lot of `AI_HUMAN_DETECTION_START` `AI_HUMAN_DETECTION_STOP` and `MOTION_STOP` but nothing appears in record folder.

I appreciate your work on this software and looking forward for any help debugging this issue.

Here is a part of output ipc_multiplexer:
```
Parsing message.
02 00 00 00 08 00 00 00 24 10 01 00 04 00 00 00 01 00 00 00
IPC message. Len: 16. Status: No error information!
Parsing message.
02 00 00 00 08 00 00 00 7b 00 01 00 00 00 00 00
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00
Sending MOTION_STOP to queue #1
IPC message. Len: 16. Status: Resource temporarily unavailable!
Parsing message.
01 00 00 00 02 00 00 00 f6 00 f6 00 00 00 00 00
IPC message. Len: 16. Status: Resource temporarily unavailable!
Parsing message.
01 00 00 00 02 00 00 00 f5 00 f5 00 00 00 00 00
Sending AI_HUMAN_DETECTION_START to queue #1
IPC message. Len: 16. Status: Resource temporarily unavailable!
Parsing message.
01 00 00 00 02 00 00 00 f6 00 f6 00 00 00 00 00
IPC message. Len: 16. Status: Resource temporarily unavailable!
Parsing message.
01 00 00 00 02 00 00 00 f5 00 f5 00 00 00 00 00
Sending AI_HUMAN_DETECTION_START to queue #1
IPC message. Len: 16. Status: Resource temporarily unavailable!
Parsing message.
01 00 00 00 02 00 00 00 f6 00 f6 00 00 00 00 00
```
Basic camera info:
```
Base Version | 8.2.0.0A_202007211802
Model Suffix | y20ga
Serial Number | 9FUS
Free SD Space | 96%
```
camera.conf:
```
WITCH_ON=yes
SAVE_VIDEO_ON_MOTION=yes
SENSITIVITY=high
AI_HUMAN_DETECTION=yes
BABY_CRYING_DETECT=yes
LED=no
ROTATE=no
IR=yes
```

roleoroleo, 2020-12-19:
The program that saves mp4 files in /tmp/record is mp4 record.
My hack doesn't change this behavior except for starting the mp4record process.
If you configure the cam in this manner:
[image]
mp4record is not started.

Adambl4, 2020-12-21:
Thank you, this is the reason of the issue, had not payed attention to that settings at all.

closed 2020-12-21 by Adambl4