roleoroleo/yi-hack-Allwinner #219: Error while decoding , bytestream -5

sermayoral, 2020-12-17:
Hi friends. I'm using this hack with Home Assistant (0.22 version), I've been watching for a while I'm getting several errors like these:

    error while decoding MB 83 52, bytestream -5
    cabac decode of qscale diff failed at 74 51
    error while decoding MB 74 51, bytestream 280
    error while decoding MB 77 62, bytestream -5
    error while decoding MB 38 53, bytestream -9

Furthermore, there are parts of the video where the image looks very bad, very pixelated.

Do you know what the problem could be?

Thanks

roleoroleo, 2020-12-17:
See this issue:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/203

sermayoral, 2020-12-17:
@roleoroleo

There was a bug. Ok. Thanks!! :-)

roleoroleo, 2020-12-27:
Did you try the beta versione of rRTSPServer posted in the issue https://github.com/roleoroleo/yi-hack-Allwinner/issues/203

closed 2021-02-15 by roleoroleo