roleoroleo/yi-hack-Allwinner #220: 9FUS 8.2.0 blinks blue no upgrade; MD5 Checksum?

Flugdino, 2020-12-18:
Hi,
after changing SD-cards ( manufactorers, sizes etc.) , formatted with Cam, Unix and MS7, the Cam won't be flashed and hangs hours in blinking blue. Was unzipped with 7zip and with an Unix OS....
After removing the SDs it works in old YI-manner. After hours of investigation I suspect an error during download. Can someone who's cam is working with the new code provide me with an MD5, please?  

[image]

My MD5 hash of 

y20ga_0.2.2.tar

is
 
5c5930304fb25926a4b301c663a3b673

generated with WinMD5Free under W7

Anyone there to confirm? Would be great!

Cheers
Peter

roleoroleo, 2020-12-19:
You have to decompress tar archive also.
Your sd must contain:
- Factory
- newhome
- home_y20gam.stage

Flugdino, 2020-12-19:
Of course, I did. Need only a point to start with the MD5....

roleoroleo, 2020-12-19:
Calculated with Hashcalc:
5c5930304fb25926a4b301c663a3b673

Flugdino, 2020-12-21:
THX a lot for the MD5 Roleo :-) You are doing a great job always :-) So, now I have to figure out whats going on with a readable SD-card and a checked downloaded image. Will now try to log via the serial what's happening...report follows. Additional ideas to solve the issue?

roleoroleo, 2020-12-22:
Are you sure the hack doesn't start?
Did you try to open the web interface http://ip:8080 ?

Flugdino, 2020-12-22:
Yes, no chance with the actual 0.2.2. direct installation. But: according to "9FUSY312E5 not flashing
#97" I tried directly the "debug_version" with with 0.1.7! 
Worked once with the Webserver than no connection, no ping, etc.
Analysing the  network traffic with Wireshark showed up some ack's of connection tries, then timed out. So SD-Card cleaned and again the "debug" with the virgin factory file installed. 
Success: 1st time flash worked and I disabled all possible number crunching staff wasting ressources of the tiny cam-cpu. 1st step was to enable RSTS. Result was still a stable connection! VLC streaming etc. works great! Reboot, PWR-off etc. of the cam doesn't influence. 
BTW the card is a SanDisk 32GB with 80MB/s. 
Next step is the in-path SW update to the 0.2.2....tell you later.  
Guess the whole trouble was caused by overloading the tiny cam cpu after reboot with a handfull processes...will crosscheck it out with the 2nd 9FUS/8.2.0 cam. 
If everyting is fine the next task ist to bring the BFUSY's with 9.19.02 on stage... keep you informed. 
Cheers Peter

closed 2021-02-15 by roleoroleo