roleoroleo/yi-hack-Allwinner #221: Since version 0.2.2 my camera reboots every 2-3 minutes

wnovak, 2020-12-28:
Firmware Version	0.2.2
Base Version	8.3.0.0A_201912271707
Model Suffix	y25ga
Serial Number	9FUSY322F81ZGJ200319
Free/Total Memory	21848/61164 KB
Free SD Space	100%

wnovak, 2020-12-28:
What do you need to analyze this issue ?

roleoroleo, 2020-12-28:
If you restore 0.2.1, does the problem disappear?
Did you change the power adapter or the usb cable?

wnovak, 2020-12-28:
restoring to 0.2.1, I not have tested this way.
and no I didn't change any power adapter or usb cable

wnovak, 2020-12-28:
OK downgraded to 0.2.1, and looks like the rebooting stays for every 2-3 Minutes

roleoroleo, 2020-12-28:
There is only 1 watchdog that reboot the cam.
It happens when rmm process dies. But I don't think that's the case.

Is there a particular configuration?

wnovak, 2020-12-28:
Should I try to remove the hack and install it new from scratch?
What you mean with particular configuration?

wnovak, 2020-12-28:
How to find out if ram process dies ?

roleoroleo, 2020-12-28:
> Should I try to remove the hack and install it new from scratch?
> What you mean with particular configuration?

Some process in cron or something started in startup.sh...

> How to find out if rmm process dies ?

Login with ssh and check repeatedly the output of the ps command or the top command.

wnovak, 2020-12-29:
OK rmm dies every 2-3 Minutes so as you say that's the reason for the reboot.

How to find out why rmm dies ?

roleoroleo, 2020-12-29:
rmm is yi program.
I don't know why it crashes.
Try to remove the hack and check if the cam works without hack.

isbkch, 2021-01-04:
I have the same issue on all my 3 cameras after I upgraded them
Here the output of `top`, not sure why the `/rmm` takes always around 35% of CPU

[image]

roleoroleo, 2021-01-05:
rmm uses 30-40 % of CPU. It's normal.
I will try to replicate the same condition with my cam.

roleoroleo, 2021-01-16:
I can't make rmm to crash.
Could you share your config?

WMP, 2021-01-26:
My camera on 0.2.2 works 90minutes and stuck, to reset i can only unplug and plug again. Its plugged to original PSU and 5m lenght cable. Do you have any debugging method to save logs on sdcard?

I thinks that this problem is related with thermal, becouse after restart works few minutes to next stuck. When i unplug it to few hours, camera can work next hour or two.

roleoroleo, 2021-01-26:
I need to know if this problem is related to the hack.
Did you try the cam before applying the hack?

WMP, 2021-01-26:
I can test on this same camera if you told me how to remove hack.

roleoroleo, 2021-01-26:
Check the wiki.

WMP, 2021-01-26:
Hmmm, i replug camera 8 hours ago and dont change anything, and still works...

roleoroleo, 2021-01-26:
Without hack?

WMP, 2021-01-26:
No, with hack in version 0.2.2. Unfortunately, going down 20 minutes ago, after 9 hours. Now i replug and test again.

cchenga, 2021-01-27:
> No, with hack in version 0.2.2. Unfortunately, going down 20 minutes ago, after 9 hours. Now i replug and test again.

Got a similar issue here. hack version 0.2.2. Stop working after (maybe) 6hours.

WMP, 2021-01-28:
Today after 11 hours my camera with yi-hack 0.2.2 reboot:
[image]

I will wait to again stuck or restart,  and then i will remove 0.2.2 hack and test without it.

WMP, 2021-01-29:
This night i have 3 lost connects, but it if very fast, with only 1 second break:

[image]

WMP, 2021-01-29:
I remove yihack, install version 0.1.8 gam, and i have problem to boot camera. I tried to reset, but i dont hear anything. I plug again sdcard with yihack remover, and my camera live. I install 0.1.8 ga and start test on this version.

roleoroleo, 2021-01-29:
Probably 1 second  break is not a problem.
The cam takes 30 seconds to reboot.
And when RTSP server crashes, takes 10 seconds to restart the daemon.
Maybe a network problem?

WMP, 2021-01-29:
Maybe, but my second camera on 0.1.8 works without any problems.

WMP, 2021-02-01:
Ok, this camera works 72h without any issue on 0.1.8 - without stuck, restart, "wifi issues". Camera works on this same place and this same USB cable and PSU. Now i will start test 0.2.0

WMP, 2021-02-03:
After 48 hours on 0.2.0 i havent any issue, now i will start test 0.2.1

WMP, 2021-02-07:
And after 96 hours with 0.2.1 i haventany issues. Now i testing 0.2.3

WMP, 2021-02-09:
On version 0.2.3 i havent  any issue since 48 hours. Now i testing again 0.2.2

WMP, 2021-02-23:
I think you can close this issue, i havent problems on 0.2.3

closed 2021-02-24 by roleoroleo