roleoroleo/yi-hack-Allwinner #225: RTSP with AAC?

andriej, 2021-01-05:
Hi,

Is there any way to enable/change PCM A-law (G711) codec of audio to AAC?
G711 is not browser compatible and because of that HomeAssistant can't stream it further with audio.

I'd like to use one of cameras as baby monitor tho.

roleoroleo, 2021-01-05:
At the moment you can choose ulaw, alaw and pcm.
I don't know if live555 supports aac. I have to check it.

andriej, 2021-01-05:
All of those are 'linear' PCM codecs at least [looking at Wikipedia](https://en.wikipedia.org/wiki/G.711) while HA and browsers have to have AAC ([source](https://developer.mozilla.org/en-US/docs/Web/Media/Formats/Audio_codecs))

roleoroleo, 2021-01-08:
The library doesn't support aac conversion.
Maybe it should be implemented using this library: https://github.com/mstorsjo/fdk-aac

andriej, 2021-01-08:
Any chance to have it implemented in coming weeks?
I can test if needed

roleoroleo, 2021-01-16:
It's a long job.
I have no time in this moment.
Sorry.

andriej, 2021-01-16:
Ok, I understand.
Could you just list points that could potentially be done, so maybe someone will want to dig further and get it implemented later on?
Thanks

roleoroleo, 2021-01-17:
Just to give it a try, I cross-compiled this:
https://github.com/mstorsjo/fdk-aac
and this:
https://github.com/nu774/fdkaac

Then I tried to run fdkaac reading form the pcm fifo and it works.
I can create an aac file.
But the hardest part is getting it into the RTSP server.

roleoroleo, 2021-01-17:
> Could you just list points that could potentially be done, so maybe someone will want to dig further and get it implemented later on?

The fdk-aac library does most of the work.
I think the simplest approach is to start from the ADTSAudioFileSource.cpp and ADTSAudioFileServerMediaSubsession.cpp files (from live555 library) and edit them to read the PCM data from the fifo and then convert it to AAC.
The fifo is in /tmp/audio_fifo and it's in the following format:
- 8 KHz
- 16 bit
- mono
- signed
- little endian

roleoroleo, 2021-01-26:
Do you want to try this alfa?
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5875613/rRTSPServer.gz)
You can start the server with the following command:
`RRTSP_RES=both RRTSP_AUDIO=aac rRTSPServer`

roleoroleo, 2021-01-26:
it was a very complicated job.
But funny.

andriej, 2021-01-26:
Sure! 

I've ran it on one of cameras I have in garage for test.
First thing I noticed was the speaker icon in the stream :-)
Then after making few noises stream broke down with no visible error in console:
```
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 167
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 172
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 173
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 260
fInargs.numInSamples 512
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 512
fOutargs.numInSamples 512
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 512
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
```

I couldn't get server back up again after that, so I rebooted that camera.
After SSH again - I did longer tests, voice seemed to be minimally out of sync from video (like before the image) but I would totally accept that and did not notice if not testing.

After like 2 mins of testing and the moment of changing IR (turned-off light), stream froze again and (I think) camera restarted as I got re-logged to SSH:
```
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 186
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 216
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 187
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 1024
fOutargs.numInSamples 1024
outargs.numOutBytes 175
fInargs.numInSamples 0
fOutargs.numInSamples 1024
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
ErrorStatus 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
outargs.numOutBytes 0
fInargs.numInSamples 0
fOutargs.numInSamples 0
ErrorStatus 0
```

andriej, 2021-01-26:
One more thing - I didn't know where to put it (as ssh is quite limited on this device) so I cilled the other server running and just ran this one from /tmp. I'm not sure if it's not other watchdog killing or something.

roleoroleo, 2021-01-26:
You can overwrite the original rRTSPServer (make a backup copy).
Then edit system.conf and set RTSP_AUDIO=aac

andriej, 2021-01-26:
Just did that and it works :-)

andriej, 2021-01-27:
I've moved the binary to the 2nd camera too - it worked, sound is there, but also I can hear like regular-distorted sound which may be somehow connected with transmission.
Regular intervals, one by one, short. I'll try to record something tommrow.

roleoroleo, 2021-01-28:
Try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5889838/rRTSPServer.gz)

andriej, 2021-01-29:
@roleoroleo [here's the example](https://imgur.com/a/3RuFBE3) recording from the camera making those sounds.
I've updated the binary on both, working on both of them - only on one those sounds are present. Probably it may be also interference, will try to move the camera tommorow and see if it helps.

Besides that - both cameras work

andriej, 2021-02-04:
I've moved camera a little and the noise is much more silent, so it may be just interference.

roleoroleo, 2021-02-06:
Fixed another bug:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5937339/rRTSPServer.gz)

andriej, 2021-02-06:
Works!

Is there any changelog tho? And will it be in main version any time soon? :-)

roleoroleo, 2021-02-06:
I didn't commit it.
But I will try to release with the next version.

andriej, 2021-04-15:
It's been a while - I guess it's within 0.2.5 already? 
Looking for some improvement regarding connectivity (debugging if it's wifi or camera itself right now).

roleoroleo, 2021-04-15:
Yes, it's released within 0.2.5.
About improvements see the release info.

closed 2024-05-12 by github-actions[bot]