roleoroleo/yi-hack-Allwinner #232: reboot after clicking snapshot url

mk386, 2021-01-11:
> # cat /proc/kmsg
> <4>[    1.830323] jffs2: Empty flash at 0x005bdffc ends at 0x005be000
> <5>[    1.830330] jffs2: CLEANMARKER node found at 0x005be000, not first node in block (0x005b0000)
> <4>[    1.830850] jffs2: Empty flash at 0x005beffc ends at 0x005bf000
> <5>[    1.830857] jffs2: CLEANMARKER node found at 0x005bf000, not first node in block (0x005b0000)
> <4>[    1.832007] jffs2: Empty flash at 0x005c0ffc ends at 0x005c1000
> <5>[    1.832015] jffs2: CLEANMARKER node found at 0x005c1000, not first node in block (0x005c0000)
> <4>[    1.832557] jffs2: Empty flash at 0x005c1ffc ends at 0x005c2000
> <5>[    1.832564] jffs2: CLEANMARKER node found at 0x005c2000, not first node in block (0x005c0000)
> <4>[    1.833094] jffs2: Empty flash at 0x005c2ffc ends at 0x005c3000
> <5>[    1.833102] jffs2: CLEANMARKER node found at 0x005c3000, not first node in block (0x005c0000)
> <4>[    1.833625] jffs2: Empty flash at 0x005c3ffc ends at 0x005c4000
> <5>[    1.833632] jffs2: CLEANMARKER node found at 0x005c4000, not first node in block (0x005c0000)
> <4>[    1.834160] jffs2: Empty flash at 0x005c4ffc ends at 0x005c5000
> <5>[    1.834168] jffs2: CLEANMARKER node found at 0x005c5000, not first node in block (0x005c0000)
> <5>[    1.834690] jffs2: CLEANMARKER node found at 0x005c6000, not first node in block (0x005c0000)
> <4>[    1.835209] jffs2: Empty flash at 0x005c6ffc ends at 0x005c7000
> <5>[    1.835216] jffs2: CLEANMARKER node found at 0x005c7000, not first node in block (0x005c0000)
> <4>[    1.835743] jffs2: Empty flash at 0x005c7ffc ends at 0x005c8000
> <5>[    1.835751] jffs2: CLEANMARKER node found at 0x005c8000, not first node in block (0x005c0000)
> <4>[    1.836269] jffs2: Empty flash at 0x005c8ffc ends at 0x005c9000
> <5>[    1.836276] jffs2: CLEANMARKER node found at 0x005c9000, not first node in block (0x005c0000)
> <4>[    1.836803] jffs2: Empty flash at 0x005c9ffc ends at 0x005ca000
> <5>[    1.836810] jffs2: CLEANMARKER node found at 0x005ca000, not first node in block (0x005c0000)
> <4>[    1.837340] jffs2: Empty flash at 0x005caffc ends at 0x005cb000
> <5>[    1.837347] jffs2: CLEANMARKER node found at 0x005cb000, not first node in block (0x005c0000)
> <4>[    1.837866] jffs2: Empty flash at 0x005cbffc ends at 0x005cc000
> <5>[    1.837873] jffs2: CLEANMARKER node found at 0x005cc000, not first node in block (0x005c0000)
> <4>[    1.838396] jffs2: Empty flash at 0x005ccffc ends at 0x005cd000
> <5>[    1.838403] jffs2: CLEANMARKER node found at 0x005cd000, not first node in block (0x005c0000)
> <4>[    1.838929] jffs2: Empty flash at 0x005cd17c ends at 0x005ce000
> <4>[    3.521252] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
> <5>[    3.582149] random: crng init done
> <4>[    3.758194] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
> <4>[    3.780747] snd_soc_sunxi: loading out-of-tree module taints kernel.
> <3>[    3.796191] --->sunxi_daudio_dev_probe<--- strat.
> <4>[    3.805609] sunxi-daudio daudio0: regulator missing or invalid
> <3>[    3.805758] --->sunxi_daudio_dev_probe<--- end.
> <6>[    3.828834] digital_vol:0, lineout_vol:31, main_gain:23, pa_msleep:160, pa_level:1
> <6>[    3.828850] adcdrc_cfg:0, adchpf_cfg:1, dacdrc_cfg:1, dachpf:0
> <4>[    3.828932] sunxi-internal-codec codec: [sunxi_internal_codec_probe] codec probe finished.
> <4>[    3.838956] [sunxi_card_init] card init finished.
> <6>[    3.839296] sunxi-codec-machine sndcodec: sun8iw19codec <-> codec mapping ok
> <4>[    3.921945] sunxi-codec-machine sndcodec: [sunxi_card_dev_probe] register card finished.
> <6>[    3.932403] snddaudio snddaudio0: codec: snd-soc-dummy, codec_dai: snd-soc-dummy-dai.
> <6>[    3.932862] snddaudio snddaudio0: snd-soc-dummy-dai <-> daudio0 mapping ok
> <4>[    3.996352] [sensor_power_warn] sensor_power_probe: cannot get sensor0_iovdd supply, setting it to NULL!
> <4>[    3.996368] [sensor_power_warn] sensor_power_probe: cannot get sensor0_avdd supply, setting it to NULL!
> <4>[    3.996378] [sensor_power_warn] sensor_power_probe: cannot get sensor0_dvdd supply, setting it to NULL!
> <6>[    4.340118] sunxi-wlan wlan: bus_index: 1
> <6>[    4.450468] sunxi-mmc sdc1: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 timing LEGACY(SDR12) dt B
> <6>[    4.469052] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
> <6>[    4.489734] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
> <6>[    4.492189] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing LEGACY(SDR12) dt B
> <3>[    4.494061] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
> <4>[    4.500301] sunxi-mmc sdc1: card claims to support voltages below defined range
> <6>[    4.509973] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
> <6>[    4.510069] sunxi-mmc sdc1: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 1 timing SD-HS(SDR25) dt B
> <6>[    4.510231] sunxi-mmc sdc1: sdc set ios:clk 50000000Hz bm PP pm ON vdd 21 width 4 timing SD-HS(SDR25) dt B
> <6>[    4.511073] mmc1: new high speed SDIO card at address 0001
> <4>[    6.110239]
> <4>[    6.110239] [CPLD_PERIPH] cpld_init ok, [ ver=Dec 27 2019, 17:08:21 ]
> <4>[    6.110681] [CPLD_PERIPH] pwm init ok.
> <4>[    6.110692] [CPLD_PERIPH] timer init ok.
> <6>[    6.110697] timer resolution:4 MHZ
> <4>[    6.110701] [CPLD_PERIPH] HZ = 250
> <4>[    8.809535] [sensor_power_warn] sensor_power_probe: cannot get sensor0_iovdd supply, setting it to NULL!
> <4>[    8.809552] [sensor_power_warn] sensor_power_probe: cannot get sensor0_avdd supply, setting it to NULL!
> <4>[    8.809562] [sensor_power_warn] sensor_power_probe: cannot get sensor0_dvdd supply, setting it to NULL!
> <4>[    9.133221] cedar_ve: VE real_fre=300000000
[... 182 lines cut ...]
> <6>[ 7530.161009] [ 1267]     0  1267      219        8       3       0        0             0 dropbear
> <6>[ 7530.161019] [ 1269]     0  1269      167        7       3       0        0             0 ipc_multiplexer
> <6>[ 7530.161028] [ 1359]     0  1359      211        7       3       0        0             0 ntpd
> <6>[ 7530.161038] [ 1380]     0  1380     1124      541       6       0        0             0 rRTSPServer
> <6>[ 7530.161049] [ 1396]     0  1396      246       10       3       0        0             0 wd_rtsp.sh
> <6>[ 7530.161059] [ 1438]     0  1438     2011       65       7       0        0             0 onvif_srvd
> <6>[ 7530.161070] [ 1443]     0  1443      518       32       5       0        0             0 wsdd
> <6>[ 7530.161080] [ 1444]     0  1444      605      444       4       0        0             0 framefinder
> <6>[ 7530.161090] [14290]     0 14290      237       26       4       0        0             0 dropbear
> <6>[ 7530.161100] [14408]     0 14408      246       10       4       0        0             0 ash
> <6>[ 7530.161111] [ 1007]     0  1007      245        9       3       0        0             0 cat
> <6>[ 7530.161121] [ 1589]     0  1589      245        8       3       0        0             0 sleep
> <6>[ 7530.161131] [ 1622]     0  1622      214        9       4       0        0             0 httpd
> <6>[ 7530.161142] [ 1623]     0  1623      245        9       3       0        0             0 snapshot.sh
> <6>[ 7530.161152] [ 1648]     0  1648     3247     1751      10       0        0             0 imggrabber
> <6>[ 7530.161161] [ 1649]     0  1649      211        6       3       0        0             0 base64
> <6>[ 7530.161171] [ 1652]     0  1652      245        8       3       0        0             0 sh
> <6>[ 7530.161182] [ 1654]     0  1654      219        6       3       0        0             0 cat
> <6>[ 7530.161192] [ 1655]     0  1655      219        6       3       0        0             0 grep
> <6>[ 7530.161202] [ 1656]     0  1656      219        6       3       0        0             0 awk
> <6>[ 7530.161212] [ 1657]     0  1657      184        9       3       0        0             0 wpa_cli
> <3>[ 7530.161218] Out of memory: Kill process 1012 (rmm) score 492 or sacrifice child
> <3>[ 7530.170837] Killed process 1012 (rmm) total-vm:43416kB, anon-rss:7240kB, file-rss:21972kB, shmem-rss:1768kB
> <6>[ 7530.235560] oom_reaper: reaped process 1012 (rmm), now anon-rss:0kB, file-rss:5676kB, shmem-rss:1764kB
> <2>[ 7532.135855] watchdog: watchdog0: watchdog did not stop!

mk386, 2021-01-11:
Info | Value
-- | --
Yi Home 1080p S/N | 9FUSY32
Firmware Version | 0.2.2
Base Version | 8.3.0.0A_201912271707
Model Suffix | y25ga

roleoroleo, 2021-01-14:
Check these issues:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/13
https://github.com/roleoroleo/yi-hack-Allwinner/issues/35

And try to enable swap space.

mk386, 2021-01-14:
Yes, it's duplicate issue. Sorry about that.

Making swap space or disabling the cloud service, both are working. I'd like to disable cloud rather than using swap on SD card.
Thank you!

closed 2021-01-14 by mk386