roleoroleo/yi-hack-Allwinner #233: rRTSPServer issue with Yi Home new firmware 9.0.19.02_202007141512

nmenoux, 2021-01-18:
I've different Yi Home cams on my project and I've encountered some issues with RTSP Server which stops working after a certain amount of time, especially at day when image changes are high. I've looked at the #204 issue and installed the last rRTSPServer.gz compiled version of rRTSPServer by @roleoroleo which nicely solved my issue.

However, when I've installed this new compiled RTSP Server on my last fresh Yi Home camera with the new firmware 9.0.19.02_202007141512, the streams is working any more, I've tested the low res.

As you can see, I've renamed the original rRTSPServer as rRTSPServer.ori and put the new compiled one.

[image]

I'm using the y21ga Allwinner-v2 firmware for this camera.

When I launch manually the server by 

`RRTSP_AUDIO=yes RRTSP_RES=both RRTSP_DEBUG=1 ./rRTSPServer`

I got this log (extract) : 

[image]

and streams is not working with VLC client.

If I re-run the original rRTSPServer.ori to get the stream back, I got this log : 

[image]

Maybe, is it due to the y21ga hack firmware for my camera or something is going wrong with stream descriptor...
I've to deep dived in the rRTSP code to find some explanations or if you have any idea @roleoroleo :-) 
Thanks a lot for your work anyway !

roleoroleo, 2021-01-18:
You can't use the binary from this hack to the Allwinner-v2 version.
The frame buffer where rRTSPServer extracts the stream is different.

nmenoux, 2021-01-18:
I suspected it indeed... is there a branch where I could find your modified rRTSPServer sources and see what could I do with the Allwinner-v2 platform ? Thanks @roleoroleo

roleoroleo, 2021-01-18:
Check the allwinner-v2 repo:
https://github.com/roleoroleo/yi-hack-Allwinner-v2

nmenoux, 2021-01-18:
are your recent changes on rRTSPServer available on the Allwinner repo ? (I mean changes to get your last rRTSPServer.gz file) I will compare them with the Allwinner-v2 src files to see if I could fix it with the new frame buffer. Thanks.

roleoroleo, 2021-01-18:
I pushed the last changes 15 minutes ago.
I ported the changes from allwinner to allwinner-v2 but I need to test them.

nmenoux, 2021-01-18:
yes I've just seen your changes @roleoroleo ;-) Thanks for this great push, hope tests will go well, I will try to compile it on my side to test on my camera.

nmenoux, 2021-01-20:
@roleoroleo could you provide the new rRTSPServer.gz for the Allwinner-v2 platform please as I'm not able to compile source files, I do not have the livemedia subdirectory for compiling. I'm compiling within a docker container but seems that I do not have all requirements for compiling. 
Thanks. Have a good day !

roleoroleo, 2021-01-20:
Please open a new issue in the v2 repo.
I don't want to upload a file here, it could be confusing.

nmenoux, 2021-01-20:
you're right @roleoroleo, I will put my request there. Thanks

closed 2024-05-12 by github-actions[bot]