roleoroleo/yi-hack-Allwinner #234: RTSP stopped working once I block the internet access of the camera

cchenga, 2021-01-27:
Hello roleoreleo,

RTSP server is working fine with internet access. However it stopped working once I blocked the internet access of the camera and restart it.
Is that expected behavior? 
Version is 0.2.2, upgraded recently.
I remembered it worked without internet access in one of the previous version. (maybe I am wrong)
Could you please confirm that?

Thanks

roleoroleo, 2021-01-27:
This is an unresolved issue.
When you block the traffic internally, shutting down the processes, it works but if you block the connection through a firewall, the cam stops working (probably a memory leak).
I don't know why, it depends on the yi programs.
https://github.com/roleoroleo/yi-hack-MStar/issues/149

Catfriend1, 2021-02-03:
Hi,

I was successful to block outgoing connections until the early days of using Yi-Hack Allwinner. Never had problems with the cam after that. There seems to be a hard-coded requirement by the cam I've discovered by trial and error: It demands access to Google DNS. 

My cam works on the local network with these rules in the firewall:
[image]

Regards,
Catfriend1

cchenga, 2021-02-04:
Thanks @Catfriend1 , it seems a good solution. But my problem is, my router doesn't allow me to add an exception to the black list. If I use white list, I need to add rules for each device in this subnet. Any suggestions?
[image]

Catfriend1, 2021-02-04:
@cchenga  Hmm maybe lettint everything pass but block http/https traffic ports 443, 80 entirely? I don't remember if there was another port which the yi cloud api used. to be sure, run e.g. wireshark when booting the cam with full internet access.

colethegamer123, 2021-02-09:
I have the same problem. Finally my whole network works in local, except the cameras what is my biggest risk and concern lets say. It would be good to do that somehow if I block Internet access with one button it works properly after.

If someone has idea what to do just tell me , im using Xiaomi Ax3600 Router.

Thank you guys :)

pomtom44, 2021-03-03:
Im having the same issue
I have logged my firewall and can see the camera trying to communicate DNS to 8.8.8.8 as well as 161.117.69.207
Does anyone know what that second IP address is?

pomtom44, 2021-03-04:
Can confirm allowing just 53 to 8.8.8.8 fixes the problem

On a side note which I dont see mentioned in this thread or the one linked before
when debugging my camera, i could see this error when I started a RTSP stream without allowing 8.8.8.8 though

ALSA lib pcm.c:8683:(pcm_write_snoop_fifo) Fifo full!

pomtom44, 2021-03-04:
Update:
For me it wasnt a network issue, (that i can see)
I have blocked all network access again for the camera, and set the following config
Disabled Cloud ON
Record without cloud off
RTSP ON
RTSP Auidio OFF
ONVIF OFF

SSH ON (Defualt I believe)
HTTPD ON
Everything else off

I cant see any traffic on my firewall any more from the camera
and my RTSP seems stable

roleoroleo, 2021-03-04:
> ALSA lib pcm.c:8683:(pcm_write_snoop_fifo) Fifo full!

This error depends on a patch I made to add audio to the stream.
Should not be a problem.

Arkady23, 2021-03-07:
Good day, roleoreleo!
I asked Andy2301 to make such a simple but high-quality RTSP. There is rarely a suitable RTSP server for old and new Yi servers.... I've read about the unsolvable difficulties that arise with direct flow. You can ask him to make such an RTPS so that he closes questions about any camera.
https://github.com/andy2301/test1/issues/1#issuecomment-792282846

ivanfor, 2021-03-11:
Hi. 
I can fully reproduce the behaviour described in this issue.
As a further analysis, I've run wireshark and seen that DNS query to 8.8.8.8 comes from /etc/resolv.conf
```
root@yi1080:~# cat /etc/resolv.conf
nameserver 192.168.1.1
nameserver 8.8.8.8
nameserver 8.208.8.243
nameserver 47.89.253.148
nameserver 47.74.247.115
```
It's querying for api.eu.xiaoyi.com. I have my own LAN DNS (192.168.1.1) and have set api.eu.xiaoyi.com to resolve to 127.0.0.1 and the RTSP server starts normally even with no internet access at all.
We can conclude that it's only needed to get an IP address for api.eu.xiaoyi.com (no matter what IP or DNS).

@roleoroleo : maybe an entry to /etc/hosts would work for everyone

Arkady23, 2021-03-11:
@ivanfor 
No, you can't. Request to api.eu.xiaoyi.com, as well as other Chinese addresses requests the "dispatch" module.

roleoroleo, 2021-03-12:
> @roleoroleo : maybe an entry to /etc/hosts would work for everyone

Many people use cloud services.
I only need to add it when the cloud is disabled.

Catfriend1, 2021-03-16:
```
echo "127.0.0.1 api.eu.xiaoyi.com" > "/home/yi-hack/etc/hosts"
mount --bind "/home/yi-hack/etc/hosts" "/etc/hosts"
# mount: mounting /home/yi-hack/etc/hosts on /etc/hosts failed: No such file or directory
```

roleoroleo, 2021-03-16:
/etc/hosts doesn't exist.
mount --bind doesn't work if the file doesn't exist.
Add this file means rewrite the whole partition.
It's not a good idea.

Catfriend1, 2021-03-17:
HOSTALIASES env var does not work, too :-/

roleoroleo, 2021-03-17:
Did you try to export it before running rmm? In system.sh

Catfriend1, 2021-03-17:
I tried to export it on she and then start udhcpd and then using wget to test. (did not work)
does rmm respect the var?

roleoroleo, 2021-03-17:
The only solution is to bind the whole folder.
```
mkdir /tmp/etc
cp -R /etc/* /tmp/etc
mount --bind /tmp/etc /etc
echo "127.0.0.1    api.eu.xiaoyi.com" > /etc/hosts
```

Catfriend1, 2021-03-17:
@roleoroleo Thought about this, too. Would you accept it when someone makes a PR like this? I fear in my stomach that could cause weird behaviour when mounting/unmounting or upgrading!?

Arkady23, 2021-03-21:
> The only solution is to bind the whole folder.
> 
> ```
> mkdir /tmp/etc
> cp -R /etc/* /tmp/etc
> mount --bind /tmp/etc /etc
> echo "127.0.0.1    api.eu.xiaoyi.com" > /etc/hosts
> ```

```
	if [ -f "$sd/record/tmp.mp4.tmp" ]; then
		killall dispatch
	fi
```

roleoroleo, 2021-03-21:
I will test this solution.

roleoroleo, 2021-03-21:
> ```
> 	if [ -f "$sd/record/tmp.mp4.tmp" ]; then
> 		killall dispatch
> 	fi
> ```

@Arkady23 
Why you need to kill dispatch?

Arkady23, 2021-03-21:
DIspatch constantly sends your network information with passwords to China. I don't like it. The camera was created for espionage purposes. It would be great if you completely remove this software.

Catfriend1, 2021-03-21:
@Arkady23  Interesting. Do you have any proof like wireshark traces or so , just out of my interest? I'm also trying to live "privacy conscious".

Arkady23, 2021-03-21:
@Catfriend1 Yes, I saw it in wireshark. The SSID and password were forwarded there. Why would they do that?!

roleoroleo, 2021-03-21:
If I remeber correctly, if dispatch is not running, rmm doesn't start.
So, we can't comment dispatch in the init script.
And kill it after some seconds, could be too late for our privacy.

Arkady23, 2021-03-21:
The situation with "RTSP stopped working once I block internet access of the camera" reminds me how web.whatsapp.com stopped working once I block internet access of the smartphone. And this does not happen with telegram.

closed 2024-05-12 by github-actions[bot]