roleoroleo/yi-hack-Allwinner #235: Yi Home 1080p Stops working after 1-2 week.

colethegamer123, 2021-01-28:
Hello Dear Developer,

I have some serious problem with the YiHack-AllWinner firmware I think so.

I have 3 Yi-Hack 1080p Camera. All of them suddenly stops working after 1-2 week of use in different time. The Camera is still pingable but cannot reach the webserver, stream stops and motion detections stop also.

All 3 Camera doing this. I unhacked all of them for testing and hacked it again with 3 version of the Firmware 
( latest 3 version I tried). Does not help, all stops after a while. Now I tried to update them to the newest again but still happened today I had to restart one of them.

I does not know the reason why this happens and before I installed the Firmware they was working of course only in Yi Home Software. 

The Cameras are the 9FUS version of Yi 1080p Home.

I can provide anything what you need just tell me. Thank you very much for your effort. Have a nice day.
Best regards,

colethegamer123, 2021-01-28:
I found SWAP answer in your earlier closed issue about same type of camera. the problem was snapshot creating. Im creating snapshot a lot of times. maybe that is the problem, im trying SWAP now, will give feedback asap. Thank you!

roleoroleo, 2021-01-28:
I think you will solve the problem using swap.

closed 2021-01-30 by colethegamer123