roleoroleo/yi-hack-Allwinner #237: No audio when user and password are set

jjtleon, 2021-01-31:
I have a couple 9FUS cameras with 8.2.0.0A_202012041528 original firmware and have been running yi-hack-Allwinner 0.2.2 successfully for a couple months now.  I really appreciate this project and its utility.  Yesterday, I upgraded to 0.2.3 and it stabilized many RTSP dropouts.  I then decided to turn on audio.  In VLC, I verified only one stream being received: video.  I tracked it down to the RTSP user authentication.  When username and password are set, no audio.  I only receive the audio stream in VLC without user authentication.

Out of curiosity, I tried the rRTSPServer from #225.  Same results with alaw, ulaw and pcm, but audio was received for aac with username and password.

Lastly, does anyone know the trick to get audio to work with the hack firmware in Blue Iris?  Thank you in advance!

roleoroleo, 2021-01-31:
I noticed the same problem but I can't understand.

Audio is correctly detected by vlc:
[image]
And decoded:
[image]

But I don't hear it .

roleoroleo, 2021-01-31:
Recompiling with the last version of live555, it seems working correctly.

jjtleon, 2021-01-31:
To be clear, the issue I am reporting is that the audio stream is missing when a username and password are required for alaw, ulaw and pcm.  When no authentication is required, audio stream is present.  Are you saying by recompiling with the latest version of live555, the no audio with user authentication issue is resolved?  Thanks in advance!

roleoroleo, 2021-01-31:
Yes.
I will send you a binary to check it.

mancmanGER, 2021-02-01:
I got the same issue, no audio in the stream. Looking forward to get the fix. THX

roleoroleo, 2021-02-01:
Try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5906030/rRTSPServer.gz)

mancmanGER, 2021-02-01:
Ok I changed the file, which does live in _/home/yi-hack/bin_ and set it to 755 via SSH
--> Sound is back

THX

jjtleon, 2021-02-01:
Confirmed audio stream is present with user authentication when RTSP_AUDIO is set to yes, alaw, ulaw, pcm and aac.  Thank you for the quick fix @roleoroleo!

setterlee, 2021-02-22:
I'm using my camera with the ispy software... WIth the Yi Home app the audio works well but with the ispy software not work... could be related with this issue?

roleoroleo, 2021-02-22:
If audio is presente using another client (eg. vlc) the problem could be:
- this issue.
- the codec is not compatible (for example my dvr doesn't recognize ulaw).
- the client doesn't work correctly.

Try with vlc and try to change the codec.

setterlee, 2021-02-28:
> If audio is presente using another client (eg. vlc) the problem could be:
> 
> * this issue.
> * the codec is not compatible (for example my dvr doesn't recognize ulaw).
> * the client doesn't work correctly.
> 
> Try with vlc and try to change the codec.

I tested with VLC and there is no sound...

nickuww, 2021-04-01:
I am using a 9FUS camera with firmware 0.2.3 and can verify that when a username and password are set there is no audio. I tested using the RTSP audio only in VLC and it will not even pull up with a username and password set. As soon as I removed them it played fine. Has anyone found a way to fix this issue?

roleoroleo, 2021-04-02:
This issue is related to the c++ compiler.
When I change the version of the library or the optimization flag the problem appear/disappear.
Try this:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6248120/rRTSPServer.gz)

closed 2021-05-15 by roleoroleo