roleoroleo/yi-hack-Allwinner #238: Yi Home 1080P BFUSY314 won't update firmware (flashing blue)

redmark330, 2021-01-31:
I've tried loading y20ga 0.2.1, 0.2.2 and 0.2.3 and only get a flashing blue led.

I formatted the MicroSD using the camera, downloaded the y20ga_0.2.x.tgz release and expanded to a folder then copied it onto the MicroSD Card but did this on a Windows 10 PC.  Inserted the MicroSD into the camera, powered it up. Got a solid yellow LED, a click and then a steady flashing blue LED. I let if run overnight but was still flashing the next morning.  Pulled the MicroSD and the camera restarts normally and can be accessed in the Yi app.

Do I need to use a Linux computer to copy the files onto the MicroSD card?

[image]
[image]
[image]

roleoroleo, 2021-01-31:
No, you don't need a linux comuter.
Try to change something about sd card:
- use another sd
- format the card with a different program (windows, linux...)

redmark330, 2021-01-31:
Thanks for the quick response...

I reformatted and tried again but no luck. I'll run out to Micro Center and pick up another MicroSD card. Just curious, what should the LED be doing while the firmware is being uploaded/installed (so I know that it's actually working) and how long should it take?

Thanks again for you help!

roleoroleo, 2021-01-31:
> I reformatted and tried again but no luck. I'll run out to Micro Center and pick up another MicroSD card. Just curious, what should the LED be doing while the firmware is being uploaded/installed (so I know that it's actually working) and how long should it take?

Sorry but I don't remember exactly the led sequence because I confuse the various models: mstar, allwinner and allwinner-v2.
To check if the hack worked, wait for blue led and try to connect to the internal web site on port 8080:
http:/IP-CAM:8080

redmark330, 2021-02-08:
Sorry for the bother but . . . I've tried 3 different brands (SanDisk, Samsung & Micro Center) MicroSD cards but still can't get any Allwinner version to load. After inserting the MicroSD card, the camera will reboot, with a solid yellow LED for about 15 seconds, then a flashing blue for 10 seconds followed by a solid blue. No web access at http://IP-CAM:8080 but can access camera through the Yi Home app on my phone. Any last suggestions/ideas before I return these cameras?

roleoroleo, 2021-02-09:
Did you try to hack the cam with another hack before Allwinner?

redmark330, 2021-02-09:
On one of the cameras I tried to install TheCrypt0/Yi-Hack-V4 but it wouldn't load either. I bought a 4-pack of the cameras and have only tried to update the firmware on 2 of them.

roleoroleo, 2021-02-09:
Try a different cam.

redmark330, 2021-02-10:
Tried remaining 2 cameras, one with SanDisk MicroSD and the other with Samsung MicroSD.  On reboot, solid yellow LED for 15-20 seconds and then steady flashing blue LED.  I'm just going to return the cameras to Micro Center and get something else. Thanks for your willingness to help with this.  I know it's not much but I'll send you another €5.00 for your time.

roleoroleo, 2021-02-10:
Don't worry.
But I would like to understand why it doesn't work.
If you want to spend more time I can help you.
For example, you could make a dump with this procedure
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)

roleoroleo, 2021-02-10:
Another question.
Are you using the provided cable and wall adaptor?
If you use a weak adaptor, often the cam doesn't start correctly.

redmark330, 2021-02-14:
Thanks for the offer. I have been using the provided cable and wall adapter. I purchased three more MicroSD cards (each a different brand) but still can't get the camera to update (still solid yellow LED to steady flashing blue).  I also tried loading the firmware backup but got the same results. What is the camera looking for in the root of the MicroSD to boot from?  Is there a hidden system file that I'm missing? That .tgz file only had a factory folder with config.sh and factory_test.sh in it.

roleoroleo, 2021-02-14:
The cam looks for a file called /Factory/factory_test.sh (in the sd card) and, if it exists, executes the file /Factory/config.sh
config.sh applies the hack and rename the file home_y20gam.stage in home_y20gam.
Then reboot the cam and load the new file system contained in home_y20gam.

roleoroleo, 2021-02-14:
What program are you using to decompress the tgz archive.
Try with 7zip if you are using Windows.

closed 2024-05-12 by github-actions[bot]