roleoroleo/yi-hack-Allwinner #239: Camera not starting with SD card after upgrading to 0.2.3 via the interface

basdejong95, 2021-02-01:
Hi, i have updated my Camera to 0.2.3 via the interface, but it won't boot (steady yellow light) until i pull the plug and remove the SD card.

IT also seemed to re-set all my settings to the defaults after the upgrade.

Firmware Version | 0.2.3
-- | --
Base Version | 8.2.0.0A_201912270941
Model Suffix | y20ga
Serial Number | BFUSY31PSIWBWxxxxxxx

roleoroleo, 2021-02-01:
If your record folder is big (a lot of mp4 files recorded) the update process needs a long time.
Try to delete the folder /tmp/sd/record and try again.

basdejong95, 2021-02-09:
> If your record folder is big (a lot of mp4 files recorded) the update process needs a long time.
> Try to delete the folder /tmp/sd/record and try again.

Thanks! that worked for me! 

Thanks for all your efforts on this!

closed 2021-02-09 by basdejong95