roleoroleo/yi-hack-Allwinner #242: Yi outdoor

UlfgarIlFabbro, 2021-02-05:
Are outdoor yi supported? is future support also foreseen with ONVIF?

roleoroleo, 2021-02-05:
It depends on the platform.
And the fw must be similar.
I don't know.

Arkady23, 2021-03-01:
1) https://github.com/shadow-1/yi-hack-v3
2) https://github.com/Arkady23/yi-hack-v3plus
3) The RTSP-server has not yet picked up this camera, because I do not have such a camera. Need help in selecting the RTSP server. If you install 1) and 2), I will offer installers from different cameras. Maybe one of them will do.

Arkady23, 2021-03-09:
:-) I will pray for you and bless you. That's as it turns out. If you do, then it's good, without ONVIF-it's also good. And Katya from Toscana said in her blog that Italy has bad programmers.... I'll write to her and tell her it's not true.
[view and mp4.tmp.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/6111943/view.and.mp4.tmp.zip)

Arkady23, 2021-03-10:
Dear roleoroleo !!!!
https://github.com/GStreamer/gst-rtsp-server
This server allows you to broadcast from an mp4 file. You also need to solve the problem of gluing mp4 and you can get rtsp for any camera.

roleoroleo, 2021-03-10:
@UlfgarIlFabbro 
Sorry, I'm getting confused.
Which platform does this issue refer to?
allwinner chip or hisilicon chip?

@Arkady23 
I already tried to use gstreamer as a RTSP server but it requires a lot of libraries and I think it's not possible to adapt it to a low resource device.

roleoroleo, 2021-03-10:
About view I wrote a simple code to extract h264 frames from the buffer.
See here:
https://github.com/alienatedsec/yi-hack-v4/tree/master/src/h264grabber/h264grabber
If the structure of the file is similar it's possible to make the code compatible.

roleoroleo, 2021-03-10:
> :-) I will pray for you and bless you. That's as it turns out. If you do, then it's good, without ONVIF-it's also good. And Katya from Toscana said in her blog that Italy has bad programmers.... [...]

Here the defines according to the file you posted:
```
// Enter here your model
#define TABLE_HIGH_OFFSET_YI_HOME_1080P 0x10
#define TABLE_LOW_OFFSET_YI_HOME_1080P 0x25A0
#define TABLE_RECORD_SIZE_YI_HOME_1080P 32
#define TABLE_RECORD_NUM_YI_HOME_1080P 300
#define BUF_SIZE_YI_HOME_1080P 781120
#define STREAM_HIGH_OFFSET_YI_HOME_1080P 0x4B40
#define STREAM_LOW_OFFSET_YI_HOME_1080P 0x81B40
#define FRAME_COUNTER_OFFSET_YI_HOME_1080P 18
#define FRAME_OFFSET_OFFSET_YI_HOME_1080P 4
#define FRAME_LENGTH_OFFSET_YI_HOME_1080P 8
```

If you build this code whit these defines it should work.
https://github.com/alienatedsec/yi-hack-v4/blob/master/src/h264grabber/h264grabber/h264grabber.c

UlfgarIlFabbro, 2021-03-10:
> hisilicon chip

cam yi outdoor old version, probably hisilicon chip

Arkady23, 2021-03-10:
many many thanks!!!!
Yi Outdoor with 1080p:

[image]

Arkady23, 2021-03-10:
I sent it view file. The size as usual is 1586752.

roleoroleo, 2021-03-10:
Maybe v3 and v4 are based on a different yi firmware version.

Arkady23, 2021-03-11:
Roleoroleo!
I don't have Linux... Could you compile grabber and rtsp-server? I'll check on my CN17 camera and two more 47US and Outdoor from the person I'm communicating with.

Arkady23, 2021-03-11:
> Maybe v3 and v4 are based on a different yi firmware version.

One of the methods of unbinding from China is installing the early firmware 1.8.6 C, which was not officially tied to China. So maybe it's better to take it as a basic one?!
For example, here https://github.com/naveednajam/17CN-Yi-Home-Recovery : Downgrading to lower version 1.8.6C to remove region blocking

UlfgarIlFabbro, 2021-03-11:
> Added: [alienatedsec/yi-hack-v4@d7bf348](https://github.com/alienatedsec/yi-hack-v4/commit/d7bf348b781055092c0fc4220ffd97a22539f0b9)

Kindly have the file (rootfs_h30 and home_h30) compiled to test if it works on yi outdoor cam model YHS-3017 6CUS?

roleoroleo, 2021-03-12:
I can share rRTSPServer and h264grabber.
I can't compile the whole solution.
[rtsp.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6130330/rtsp.tar.gz)

UlfgarIlFabbro, 2021-03-12:
How do I insert these files?  Just extract the folder and put it in the yi outdoor coupon together with the yi hack 4?

Come inserisco questi file? Basta estrarre la cartella e metterla nella schedina della yi outdoor insieme alla yi hack 4?

roleoroleo, 2021-03-12:
Just for testing, you can copy files to /tmp or to the sd card and run them.
```
./h264grabber -r high -m MODEL -f &
RRTSP_RES=high ./rRTSPServer &
```

Arkady23, 2021-03-13:
Dear @roleoroleo !!!
I've checked on CN17 so far. I saw the picture in VLC 2, maybe 3 seconds. And then everything went out. I went to telnet and checked the operation of the first module. Perhaps this is how it should be, as in the screenshot below.
[image]

roleoroleo, 2021-03-13:
Sorry but I don't know how to help you.
I don't own this model.

Arkady23, 2021-03-13:
RTSP2303 works for about 1 minute.
startOFS=19264 repeatOFS=399980 readsPMON=-1 ./rtsp2303 &

Arkady23, 2021-03-13:
> I can share rRTSPServer and h264grabber.
> I can't compile the whole solution.
> [rtsp.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6130330/rtsp.tar.gz)

I tried your module on the yi-hack-v3 firmware. Maybe try it on yi-hack-v4? But then you need to recompile to the length of view 648000.

Arkady23, 2021-03-18:
> Sorry but I don't know how to help you.
> I don't own this model.

@roleoroleo !!! Another question for you. Why do files always have zero length? Because they're always busy and don't show up?
[image]

roleoroleo, 2021-03-18:
They are not normal files, they are fifo.

roleoroleo, 2021-03-22:
@Arkady23 
https://github.com/alienatedsec/yi-hack-v4/commit/2a6d5bb38bd6afec42b7c8250a1eb75527250559

Arkady23, 2021-03-22:
@roleoroleo  I believe you. And make me a module to test! I sent the last module to the user with Outdoor 1080p. But he doesn't answer. He must have gone somewhere.

roleoroleo, 2021-03-22:
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6180816/h264grabber.gz)

Arkady23, 2021-03-22:
@roleoroleo You're a genius! I want to note that you were the first to make this program. There was no rtsp for these cameras for 5 years. All over the Internet, only this is written that there is no rtsp for this camera and compatible models. This is a success.
I've seen that only your rtsp-server option works on Al+. That's why I came to you. Only rtsp-high works on yi-hack-v3. Because you put the addresses on the source 648000, as on yi-hack-v4. If you consider it necessary to provide for the yi_home_v3 option, then it would be nice. If not, then there is no trial. I will check your module on v4 soon. And I will write.

Arkady23, 2021-03-22:
Perfectly! Both work as planned.

roleoroleo, 2021-03-22:
@Arkady23 
Do you want to try this binary?
[imggrabber.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6185101/imggrabber.tar.gz)

Arkady23, 2021-03-23:
This thing doesn't work: `/home/yi-hack/www/cgi-bin/imggrabber-r high > /tmp/sd/ark. png`. The output file is zero.

roleoroleo, 2021-03-23:
Add -m yi_home

Arkady23, 2021-03-23:
Only low works. High mode takes a long time to think and outputs a null file.

roleoroleo, 2021-03-23:
Please, try to run it with "-d" and post the output.

Arkady23, 2021-03-23:
Empty again.

roleoroleo, 2021-03-23:
Nothing in console?

Arkady23, 2021-03-23:
```
/home/yi-hack-v4 # /home/app/imggrabber -m yi_home -r high -d > /tmp/sd/ark.png
Debug on
Starting program
Resolution high
1616534658814 - mapping file /tmp/view, size 648000, to b6e08000
1616534658828 - closing the file /tmp/view
1616534658829 - found latest frame: id 36, frame_counter 12785
1616534658831 - processing frame 36
1616534658859 - processing frame 36
1616534658880 - processing frame 36
1616534658908 - processing frame 36
1616534658928 - processing frame 36
1616534658951 - processing frame 37
1616534658978 - processing frame 37
1616534659004 - processing frame 38
1616534659028 - processing frame 38
1616534659050 - processing frame 39
1616534659078 - processing frame 39
1616534659101 - processing frame 40
1616534659128 - processing frame 40
1616534659149 - processing frame 41
1616534659181 - processing frame 41
1616534659208 - processing frame 42
1616534659228 - processing frame 42
1616534659250 - processing frame 43
1616534659278 - processing frame 43
1616534659304 - processing frame 44
1616534659328 - processing frame 44
1616534659350 - processing frame 45
1616534659378 - processing frame 45
1616534659402 - processing frame 46
1616534659428 - processing frame 46
1616534659450 - processing frame 47
1616534659478 - processing frame 47
1616534659504 - processing frame 48
1616534659528 - processing frame 48
1616534659553 - processing frame 49
1616534659582 - processing frame 49
1616534659608 - processing frame 50
1616534659628 - processing frame 50
1616534659652 - processing frame 51
1616534659680 - processing frame 51
1616534659708 - processing frame 52
1616534659728 - processing frame 52
1616534659752 - processing frame 53
1616534659779 - processing frame 53
1616534659805 - processing frame 54
1616534659828 - processing frame 54
1616534659852 - processing frame 55
1616534659878 - processing frame 55
1616534659901 - processing frame 56
1616534659929 - processing frame 56
1616534659953 - processing frame 57
1616534659979 - processing frame 57
1616534660005 - processing frame 58
1616534660029 - processing frame 58
1616534660053 - processing frame 59
1616534660079 - processing frame 59
1616534660101 - processing frame 60
1616534660101 - writing frame: frame_offset 68688, frame_ptr b6e1d790, frame_length 35
1616534660129 - processing frame 61
1616534660129 - writing frame: frame_offset 68728, frame_ptr b6e1d7b8, frame_length 8
1616534660153 - processing frame 62
1616534660153 - writing frame: frame_offset 68740, frame_ptr b6e1d7c4, frame_length 62034
7392875667064784 - frame found, exit loop
Encoding h264 frame
Killed
/home/yi-hack-v4 #
```

Arkady23, 2021-03-23:
I repeated it again:
```
/home/yi-hack-v4 # /home/app/imggrabber -m yi_home -r high -d > /tmp/sd/ark.png
Debug on
Starting program
Resolution high
1616534821695 - mapping file /tmp/view, size 648000, to b6e2e000
1616534821699 - closing the file /tmp/view
1616534821700 - found latest frame: id 1, frame_counter 0
1616534821706 - processing frame 1
1616534821737 - processing frame 1
1616534821757 - processing frame 1
1616534821777 - processing frame 1
1616534821797 - processing frame 1
1616534821818 - processing frame 1
1616534821837 - processing frame 1
1616534821858 - processing frame 1
1616534821877 - processing frame 1
1616534821898 - processing frame 1
1616534821918 - processing frame 1
1616534821937 - processing frame 1
1616534821958 - processing frame 1
1616534821977 - processing frame 1
1616534821997 - processing frame 1
1616534822018 - processing frame 1
1616534822038 - processing frame 1
1616534822057 - processing frame 1
1616534822077 - processing frame 1
1616534822097 - processing frame 1
1616534822118 - processing frame 1
1616534822147 - processing frame 1
1616534822169 - processing frame 1
1616534822197 - processing frame 1
1616534822218 - processing frame 1
1616534822239 - processing frame 1
1616534822269 - processing frame 1
1616534822297 - processing frame 1
1616534822318 - processing frame 1
1616534822337 - processing frame 1
1616534822358 - processing frame 1
1616534822387 - processing frame 1
1616534822407 - processing frame 1
1616534822430 - processing frame 1
1616534822458 - processing frame 1
1616534822478 - processing frame 1
1616534822498 - processing frame 1
1616534822518 - processing frame 1
1616534822538 - processing frame 1
1616534822558 - processing frame 1
1616534822578 - processing frame 1
1616534822597 - processing frame 1
1616534822618 - processing frame 1
1616534822637 - processing frame 1
1616534822657 - processing frame 1
1616534822679 - processing frame 1
1616534822698 - processing frame 1
1616534822719 - processing frame 1
1616534822748 - processing frame 1
1616534822768 - processing frame 1
1616534822791 - processing frame 1
1616534822818 - processing frame 1
1616534822838 - processing frame 1
1616534822858 - processing frame 1
1616534822878 - processing frame 1
1616534822898 - processing frame 1
1616534822918 - processing frame 1
1616534822939 - processing frame 1
1616534822958 - processing frame 1
1616534822978 - processing frame 1
1616534822998 - processing frame 1
1616534823018 - processing frame 1
1616534823038 - processing frame 1
1616534823058 - processing frame 1
1616534823078 - processing frame 1
1616534823098 - processing frame 1
1616534823118 - processing frame 1
1616534823138 - processing frame 1
1616534823158 - processing frame 1
1616534823178 - processing frame 1
1616534823199 - processing frame 1
1616534823228 - processing frame 1
1616534823248 - processing frame 1
1616534823268 - processing frame 1
1616534823290 - processing frame 1
1616534823318 - processing frame 1
1616534823338 - processing frame 1
1616534823358 - processing frame 1
1616534823378 - processing frame 1
1616534823398 - processing frame 1
1616534823419 - processing frame 1
1616534823438 - processing frame 1
1616534823458 - processing frame 1
1616534823478 - processing frame 1
1616534823498 - processing frame 1
1616534823519 - processing frame 1
1616534823538 - processing frame 1
1616534823558 - processing frame 1
1616534823578 - processing frame 1
1616534823599 - processing frame 1
1616534823619 - processing frame 1
1616534823638 - processing frame 1
1616534823659 - processing frame 1
1616534823678 - processing frame 1
1616534823698 - processing frame 1
1616534823719 - processing frame 2
1616534823739 - processing frame 2
1616534823758 - processing frame 3
1616534823778 - processing frame 3
1616534823798 - processing frame 3
1616534823819 - processing frame 4
1616534823838 - processing frame 4
1616534823858 - processing frame 4
1616534823878 - processing frame 5
1616534823898 - processing frame 5
1616534823919 - processing frame 6
1616534823938 - processing frame 6
1616534823959 - processing frame 6
1616534823989 - processing frame 7
1616534824008 - processing frame 8
1616534824031 - processing frame 8
1616534824058 - processing frame 8
1616534824078 - processing frame 9
1616534824098 - processing frame 9
1616534824119 - processing frame 10
1616534824120 - writing frame: frame_offset 388616, frame_ptr b6e91948, frame_length 35
1616534824148 - processing frame 11
1616534824148 - writing frame: frame_offset 388656, frame_ptr b6e91970, frame_length 8
1616534824169 - processing frame 12
1616534824170 - writing frame: frame_offset 4, frame_ptr b6e32b44, frame_length 58755
7392875667064784 - frame found, exit loop
Encoding h264 frame
Killed
/home/yi-hack-v4 #
```

Arkady23, 2021-03-23:
And more:
```
/home/yi-hack-v4 # /home/app/imggrabber -m yi_home -r high -w -d > /tmp/sd/ark.png
Debug on
Starting program
Resolution high
1616535286837 - mapping file /tmp/view, size 648000, to b6d7d000
1616535286840 - closing the file /tmp/view
1616535286842 - found latest frame: id 70, frame_counter 25869
1616535286844 - processing frame 70
1616535286875 - processing frame 70
1616535286895 - processing frame 70
1616535286915 - processing frame 70
1616535286936 - processing frame 70
1616535286958 - processing frame 71
1616535286986 - processing frame 71
1616535287016 - processing frame 72
1616535287036 - processing frame 72
1616535287058 - processing frame 73
1616535287086 - processing frame 73
1616535287115 - processing frame 74
1616535287136 - processing frame 74
1616535287158 - processing frame 75
1616535287186 - processing frame 75
1616535287214 - processing frame 76
1616535287236 - processing frame 76
1616535287258 - processing frame 77
1616535287286 - processing frame 77
1616535287316 - processing frame 78
1616535287336 - processing frame 78
1616535287359 - processing frame 79
1616535287388 - processing frame 79
1616535287416 - processing frame 80
1616535287446 - processing frame 80
1616535287466 - processing frame 81
1616535287486 - processing frame 81
1616535287516 - processing frame 82
1616535287536 - processing frame 82
1616535287559 - processing frame 83
1616535287586 - processing frame 83
1616535287611 - processing frame 84
1616535287636 - processing frame 84
1616535287658 - processing frame 85
1616535287686 - processing frame 85
1616535287716 - processing frame 86
1616535287736 - processing frame 86
1616535287759 - processing frame 87
1616535287787 - processing frame 87
1616535287816 - processing frame 88
1616535287846 - processing frame 88
1616535287866 - processing frame 89
1616535287886 - processing frame 89
1616535287916 - processing frame 90
1616535287936 - processing frame 90
1616535287966 - processing frame 91
1616535287986 - processing frame 91
1616535288011 - processing frame 92
1616535288036 - processing frame 92
1616535288059 - processing frame 93
1616535288086 - processing frame 93
1616535288116 - processing frame 94
1616535288116 - writing frame: frame_offset 23296, frame_ptr b6d87640, frame_length 35
1616535288138 - processing frame 95
1616535288139 - writing frame: frame_offset 23336, frame_ptr b6d87668, frame_length 8
1616535288166 - processing frame 96
1616535288166 - writing frame: frame_offset 23348, frame_ptr b6d87674, frame_length 57903
7392875667064784 - frame found, exit loop
Encoding h264 frame
Killed
/home/yi-hack-v4 #
```

roleoroleo, 2021-03-23:
Ok.
The frames are correctly detected.
But are not converted.
I will check it.
Is the resolution 1280x720?

Arkady23, 2021-03-23:
720p-that's for sure.

roleoroleo, 2021-03-24:
It's probably a memory problem.
This binary uses libavcodec and it's very big in term of memory used.
How long do you wait before the break?

Arkady23, 2021-03-24:
I didn't exactly measure it. I think about 10 seconds. I'll check it out.

Arkady23, 2021-03-25:
Ok. I left only http and telnet, disabled everything else. There is now 6776 free memory. The output file is empty.
```
 -----------------------------------------------------
  yi-hack-v4 - 0.2.0
 -----------------------------------------------------

/home/yi-hack-v4 # /home/yi-hack-v4/bin/imggrabber -m yi_home -r high > /tmp/sd/ark.png
Resolution high
/home/yi-hack-v4 # /home/yi-hack-v4/bin/imggrabber -m yi_home -r high -d > /tmp/sd/ark.png
Debug on
Starting program
Resolution high
1614865737062 - mapping file /tmp/view, size 648000, to b6d7f000
1614865737068 - closing the file /tmp/view
1614865737075 - found latest frame: id 52, frame_counter 6351
1614865737078 - processing frame 52
1614865737104 - processing frame 52
1614865737123 - processing frame 52
1614865737142 - processing frame 52
1614865737162 - processing frame 53
1614865737182 - processing frame 53
1614865737202 - processing frame 53
1614865737223 - processing frame 54
1614865737242 - processing frame 54
1614865737262 - processing frame 55
1614865737282 - processing frame 55
1614865737302 - processing frame 55
1614865737324 - processing frame 56
1614865737342 - processing frame 56
1614865737362 - processing frame 57
1614865737383 - processing frame 57
1614865737402 - processing frame 57
1614865737423 - processing frame 58
1614865737442 - processing frame 58
1614865737462 - processing frame 59
1614865737483 - processing frame 59
1614865737502 - processing frame 59
1614865737524 - processing frame 60
1614865737542 - processing frame 60
1614865737562 - processing frame 61
1614865737582 - processing frame 61
1614865737602 - processing frame 61
1614865737623 - processing frame 62
1614865737642 - processing frame 62
1614865737662 - processing frame 63
1614865737682 - processing frame 63
1614865737702 - processing frame 63
1614865737723 - processing frame 64
1614865737742 - processing frame 64
1614865737762 - processing frame 65
1614865737782 - processing frame 65
1614865737802 - processing frame 65
1614865737823 - processing frame 66
1614865737842 - processing frame 66
1614865737862 - processing frame 67
1614865737882 - processing frame 67
1614865737902 - processing frame 67
1614865737923 - processing frame 68
1614865737943 - processing frame 68
1614865737962 - processing frame 69
1614865737982 - processing frame 69
1614865738002 - processing frame 69
1614865738023 - processing frame 70
1614865738043 - processing frame 70
1614865738062 - processing frame 71
1614865738082 - processing frame 71
1614865738103 - processing frame 71
1614865738123 - processing frame 72
1614865738124 - writing frame: frame_offset 193420, frame_ptr b6db2ecc, frame_length 35
1614865738142 - processing frame 73
1614865738142 - writing frame: frame_offset 193460, frame_ptr b6db2ef4, frame_length 8
1614865738162 - processing frame 74
1614865738162 - writing frame: frame_offset 193472, frame_ptr b6db2f00, frame_length 57845
7392875667064784 - frame found, exit loop
Encoding h264 frame
Encoding jpeg image
Unmapping file /tmp/view, size 648000, from b6d7f000
/home/yi-hack-v4 #
```

Arkady23, 2021-03-25:
I did some experiments on memory loading. I even had imggrabber build a high drawing for about five minutes, but built it. I also tend to think that the problem is insufficient RAM.

Arkady23, 2021-03-25:
I am watching the recording on the SD on the yi-hack-v4 firmware. The recording goes on and on. The reason is unclear. I don't do anything with the camera. Just watching the /tmp/sd/record/ folder. I want to go back to yi-hack-v3. There is no such thing. I don't like this situation. The picture below shows how the recording is constantly interrupted, and then it resumes itself.
[image]

Arkady23, 2021-03-27:
It's a pity that RTSP-low/SD for view 781120 doesn't work. I checked ipc_cmd for CN17 — it doesn't work too.

Arkady23, 2021-03-27:
It's a pity that RTSP-low/SD for view 781120 doesn't work. I checked ipc_cmd for CN17 — it doesn't work too.
Dear @roleoroleo !!! I have just received an email from the owner of the Outdoor camera. He said that your new RTSP on his Outdoor does not work. Dear Roleo, maybe instead of the camera model key -m, set the parameters of the view file? Something like this:
```
table_offset=9999 table_record_size=9999 table_record_num=9999 ./h264grabber &
```
And then this program will be universal. You can set values for any case. And even pick up for some special option.

roleoroleo, 2021-03-27:
I will add a configuration file to the program.

roleoroleo, 2021-03-27:
Added 10 command line arguments.
Use "-m custom" and set the following parameters:

```
--table_high_offset
offset of the high res table (normally is 10)
--table_low_offset
offset of the low res table (it depends on the model)
--table_record_size
size of the record in the table (normally is 32)
--table_record_num
number of record in the table (it depends on the model)
--buf_size
size of the buffer file (it depends on the model)
--stream_high_offset
offset of the high res stream (it depends on the model)
--stream_low_offset
offset of the low res stream (it depends on the model)
--frame_counter_offset
offset of the frame counter in the record (normally is 18)
--frame_offset_offset
offset of the frame offset in the record (normally is 4)
--frame_length_offset
offset of the frame lenght in the record (normally is 8)
```
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6215588/h264grabber.gz)

Arkady23, 2021-03-27:
I mean, the number of parameters is redundant. You can't set them together (low and high). Then it would be more beautiful to remove the part and use it together with the-r key. For example:
```
h264grabber -r low -f -m custom --table_offset 4832 --table_record_num 150 --buf_size 648000 --stream_offset 428864
```
or
```
 h264grabber -r high -f -m custom --table_offset 16 --table_record_num 150 --buf_size 648000 --stream_offset 19264
```

roleoroleo, 2021-03-27:
Ok.

Arkady23, 2021-03-28:
Many-many-many thanks.
I found the address where the low stream starts for the file buf_size=781120 — stream_offset=531264.

Arkady23, 2021-03-28:
It seems to me that something is not working right.
I start the process with this command:
```
./h264grabber -r low -m custom -f --buf_size 781120 --stream_offset 531264 --table_offset 4832 --table_record_num 150 -d &
```
The program outputs:
```
Welcome to HiLinux.
~ # cd /home/app
/home/app # ./h264grabber -r low -m custom -f --buf_size 781120 --stream_offset 531264 --table_offset 4832 --table_record_num 150 -d &
/home/app # Using fifo as output
Debug on

Usage: ./h264grabber [-r RES] [-d]

        -r RES, --resolution RES
                set resolution: LOW or HIGH (default HIGH)
        -m MODEL, --model MODEL
                select cam model: yi_home, yi_home_1080, yi_dome_720p, yi_outdoor or custom
        --table_offset
                offset of the table for the resolution selected
        --table_record_size
                size of the record in the table
        --table_record_num
                number of record in the table
        --buf_size
                size of the buffer file
        --stream_offset
                offset of the stream for the resolution selected
        --frame_counter_offset
                offset of the frame counter in the record
        --frame_offset_offset
                offset of the frame offset in the record
        --frame_length_offset
                offset of the frame lenght in the record
        -f, --fifo
                enable fifo output
        -d, --debug
                enable debug
        -h, --help
                print this help
```
I don't set any other options, because they are the same by default as all the others. What's wrong?

roleoroleo, 2021-03-28:
You have to enter all the following parameters:
```
--table_high_offset
offset of the high res table (normally is 10)
--table_low_offset
offset of the low res table (it depends on the model)
--table_record_size
size of the record in the table (normally is 32)
--table_record_num
number of record in the table (it depends on the model)
--buf_size
size of the buffer file (it depends on the model)
--stream_high_offset
offset of the high res stream (it depends on the model)
--stream_low_offset
offset of the low res stream (it depends on the model)
--frame_counter_offset
offset of the frame counter in the record (normally is 18)
--frame_offset_offset
offset of the frame offset in the record (normally is 4)
--frame_length_offset
offset of the frame lenght in the record (normally is 8)
```
There is no default.

Arkady23, 2021-03-28:
You can see in the message above I wrote to you that I specifically set all 8 options for the option that should work 100%!.... And nothing has changed.... it also doesn't work.

./h264grabber -r high -m custom -f --buf_size 648000 --stream_offset 19264 --table_offset 16 --table_record_num 150 --table_record_size 32 --frame_counter_offset 18 --frame_offset_offset 4 --frame_length_offset 8 &

Arkady23, 2021-03-29:
You can see in the message above I wrote to you that I specifically set all 8 (eight) options for the option that should work 100%!.... And nothing has changed.... it also doesn't work:

./h264grabber -r high -m custom -f --buf_size 648000 --stream_offset 19264 --table_offset 16 --table_record_num 150 --table_record_size 32 --frame_counter_offset 18 --frame_offset_offset 4 --frame_length_offset 8 &

And why can't it be done by default option?. There in the src it was! And don't need any "custom". It's just that these additional parameters need to be corrected. I see this text in C like this:
```
        switch (c) {
        case 'r':
            if (strcasecmp("low", optarg) == 0) {
                resolution = RESOLUTION_LOW;
            } else if (strcasecmp("high", optarg) == 0) {
                resolution = RESOLUTION_HIGH;
            }
            break;

        case 'm':
            if (strcasecmp("yi_home", optarg) == 0) {
                table_high_offset = TABLE_HIGH_OFFSET_YI_HOME;
                table_low_offset = TABLE_LOW_OFFSET_YI_HOME;
                table_record_size = TABLE_RECORD_SIZE_YI_HOME;
                table_record_num = TABLE_RECORD_NUM_YI_HOME;
                buf_size = BUF_SIZE_YI_HOME;
                stream_high_offset = STREAM_HIGH_OFFSET_YI_HOME;
                stream_low_offset = STREAM_LOW_OFFSET_YI_HOME;
                frame_counter_offset = FRAME_COUNTER_OFFSET_YI_HOME;
                frame_offset_offset = FRAME_OFFSET_OFFSET_YI_HOME;
                frame_length_offset = FRAME_LENGTH_OFFSET_YI_HOME;
            } else if (strcasecmp("yi_home_1080p", optarg) == 0) {
                table_high_offset = TABLE_HIGH_OFFSET_YI_HOME_1080P;
                table_low_offset = TABLE_LOW_OFFSET_YI_HOME_1080P;
                table_record_size = TABLE_RECORD_SIZE_YI_HOME_1080P;
                table_record_num = TABLE_RECORD_NUM_YI_HOME_1080P;
                buf_size = BUF_SIZE_YI_HOME_1080P;
                stream_high_offset = STREAM_HIGH_OFFSET_YI_HOME_1080P;
                stream_low_offset = STREAM_LOW_OFFSET_YI_HOME_1080P;
                frame_counter_offset = FRAME_COUNTER_OFFSET_YI_HOME_1080P;
                frame_offset_offset = FRAME_OFFSET_OFFSET_YI_HOME_1080P;
                frame_length_offset = FRAME_LENGTH_OFFSET_YI_HOME_1080P;
            } else if (strcasecmp("yi_dome_720p", optarg) == 0) {
                table_high_offset = TABLE_HIGH_OFFSET_YI_DOME_720P;
                table_low_offset = TABLE_LOW_OFFSET_YI_DOME_720P;
                table_record_size = TABLE_RECORD_SIZE_YI_DOME_720P;
                table_record_num = TABLE_RECORD_NUM_YI_DOME_720P;
                buf_size = BUF_SIZE_YI_DOME_720P;
                stream_high_offset = STREAM_HIGH_OFFSET_YI_DOME_720P;
                stream_low_offset = STREAM_LOW_OFFSET_YI_DOME_720P;
                frame_counter_offset = FRAME_COUNTER_OFFSET_YI_DOME_720P;
                frame_offset_offset = FRAME_OFFSET_OFFSET_YI_DOME_720P;
                frame_length_offset = FRAME_LENGTH_OFFSET_YI_DOME_720P;
            } else if (strcasecmp("yi_outdoor", optarg) == 0) {
                table_high_offset = TABLE_HIGH_OFFSET_YI_OUTDOOR;
                table_low_offset = TABLE_LOW_OFFSET_YI_OUTDOOR;
                table_record_size = TABLE_RECORD_SIZE_YI_OUTDOOR;
                table_record_num = TABLE_RECORD_NUM_YI_OUTDOOR;
                buf_size = BUF_SIZE_YI_OUTDOOR;
                stream_high_offset = STREAM_HIGH_OFFSET_YI_OUTDOOR;
                stream_low_offset = STREAM_LOW_OFFSET_YI_OUTDOOR;
                frame_counter_offset = FRAME_COUNTER_OFFSET_YI_OUTDOOR;
                frame_offset_offset = FRAME_OFFSET_OFFSET_YI_OUTDOOR;
                frame_length_offset = FRAME_LENGTH_OFFSET_YI_OUTDOOR;
            }
            break;

        case 'table_offset':
            table_offset2 = c;
            break;
        case 'table_record_size':
            table_record_size2 = c;
            break;
        case 'table_record_num':
[... 54 lines cut ...]
        }
    }
    if (table_record_size2 > 0 ) {
        table_record_size = table_record_size2
    }
    if (table_record_num2 > 0 ) {
        table_record_num2 = table_record_num2
    }
    if (buf_size2 > 0 ) {
        buf_size = buf_size2
    }
    if (frame_counter_offset2 > 0 ) {
        frame_counter_offset = frame_counter_offset2
    }
    if (frame_offset_offset2 > 0 ) {
        frame_offset_offset = frame_offset_offset2
    }
    if (frame_length_offset2 > 0 ) {
        frame_length_offset = frame_length_offset2
    }
```
I have never written in C, so I think that we still need to add some function to convert the type of the variable "c". And first describe these new 8 variables ending in "2" as int.

And the program call in my case will be like this:
`./h264grabber -r low -m yi_home -f stream_offset 531264 &`

roleoroleo, 2021-03-29:
Try this:
[h264grabber.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6219752/h264grabber.gz)

Arkady23, 2021-03-29:
```
/home/app # ./h264grabber -h

Usage: ./h264grabber [-r RES] [-d]

        -r RES, --resolution RES
                set resolution: LOW or HIGH (default HIGH)
        -m MODEL, --model MODEL
                select cam model: yi_home, yi_home_1080, yi_dome_720p, yi_outdoor or custom
        --table_offset
                offset of the table for the resolution selected
        --table_record_size
                size of the record in the table
        --table_record_num
                number of record in the table
        --buf_size
                size of the buffer file
        --stream_offset
                offset of the stream for the resolution selected
        --frame_counter_offset
                offset of the frame counter in the record
        --frame_offset_offset
                offset of the frame offset in the record
        --frame_length_offset
                offset of the frame lenght in the record
        -f, --fifo
                enable fifo output
        -d, --debug
                enable debug
        -h, --help
                print this help
/home/app # ./h264grabber -r low -m yi_home --srteam_offset 531264 -f &
/home/app # ./h264grabber: unrecognized option `--srteam_offset'
Using fifo as output
Resolution low
^C
/home/app # ./h264grabber -r low -m yi_home -srteam_offset 531264 -f &
/home/app # ./h264grabber: invalid option -- s
Using fifo as output
Resolution low
^C
/home/app # ./h264grabber -r low -m custom --srteam_offset 531264 --buf_size 781120 --table_offset 4832 --table_record_num 150 -f &
/home/app # ./h264grabber: unrecognized option `--srteam_offset'
Using fifo as output
Resolution low
^C
/home/app # ./h264grabber -r low -m custom --srteam_low_offset 531264 --buf_size 781120 --table_offset 4832 --table_record_num 150 -f &
/home/app # ./h264grabber: unrecognized option `--srteam_low_offset'
Using fifo as output
Resolution low
^C
/home/app #
```

roleoroleo, 2021-03-29:
Check the syntax.

Arkady23, 2021-03-29:
oh!!!
```
if [ "$RTSP" == "yes" ] ; then
	if [ "$L" == "yes" ] ; then
	  N=$(ls -l /tmp/view | awk '{ print $5 }')
	  if [ $N == 648000 ] ; then
		./h264grabber -r low -m $cm -f &
	  elif [ $N == 781120 ] ; then
		./h264grabber -r low -m $cm -f --buf_size $N --stream_offset 531264 &
	  fi
	fi
	./h264grabber -r high -m $cm -f &
	sleep 1
	./rRTSPServer -r both &
fi
```
[image]

[2 more replies]

closed 2021-05-15 by roleoroleo