roleoroleo/yi-hack-Allwinner #245: Web Service - Record: Not properly working

jorgedeand, 2021-02-24:
Hello,

The issue is related to this camera:
Yi 1080p Home BFUS - Version: 8.2.0.0A_202012041528
yi-hack-Allwinner - 0.2.3

When I use the Web Service Record it begins to record but never stops. It keeps recording videos of 1 minute long until I stop it by rebooting the camera or by service mp4record:stop

I use the following address, with mi internal IP.
http://IP:8080/cgi-bin/record.sh?time=TIME

For TIME I have tried: 1, 10, 30, 60, 90...
With each one, it begins to record videos one after another.
Are there some specific valid values that I am not aware of?

I think it is not related to this version, 0.2.3, because it had happened to me before with earlier versions.
It is also happening to another camera I have with Alwinner-v2. If you want I can open a similar issue for it in the Allwinner-v2 issues.

roleoroleo, 2021-02-24:
Try these new binaries:
[ipc_progs.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6038606/ipc_progs.tar.gz)

jorgedeand, 2021-02-25:
I have replaced the binaries in folder yi-hack/bin using scp

It is working now! Will you add it for next release?

Will these binary files work for yi-hack-v2 too?  

Thank you!

roleoroleo, 2021-02-25:
> It is working now! Will you add it for next release?

Yes.
 
> Will these binary files work for yi-hack-v2 too?

No.

ricardojlrufino, 2021-03-23:
> Try these new binaries:
> [ipc_progs.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6038606/ipc_progs.tar.gz)

is this in this version ? yi-hack-Allwinner 0.2.4

roleoroleo, 2021-03-24:
Yes.
https://github.com/roleoroleo/yi-hack-Allwinner/commit/bcaf6795d3e5c5bac27e92573d6daac4b3a48af7

closed 2024-05-11 by github-actions[bot]