roleoroleo/yi-hack-Allwinner #248: Problem Hacking camera

pomtom44, 2021-03-02:
Hi team

Im looking for some help
I have 2 cameras, (9FUS) which im trying to hack, but no matter what I do I cant seem to get it to work

I first tried to put the hack on a SD card with the wifi config settings but that didnt seem to work 
I just got a orange light flasing looking like it was waiting to connect, but the console output didnt show anything about it updating

So then I tried to reset the camera and set it up as new, using theapp, then put the hack on, but it didnt seem to apply (web or rtsp didnt enable)

Im sure im doing something wrong but i dont know what
Can someone help me with troubleshooting

I have serial access to one of the cameras so I can do anything there which you need
and have copies of the logs from my attempts before if those are felpful

Thanks in advance

roleoroleo, 2021-03-03:
If you have a lot of mp4 files recorded in your sd card, the hack takes a long time.
Remove record folder from sd and try again.

pomtom44, 2021-03-03:
> If you have a lot of mp4 files recorded in your sd card, the hack takes a long time.
> Remove record folder from sd and try again.

Thanks for the comment Roleo
This is a new camera with a new SD card so there isnt any recordings currently

I have had a look though the log for when i tried the hack the first time and can see the following lines which seem note worthy

`############## Starting Hack ##############
### Hacking telnet
Enabling telnet
### Hacking yi-hack script
Enabling yi-hack script
### Disabling Yi Junk
### Updating /home/base/tools/extpkg.sh
### Disabling hack for next reboot
/tmp/sd/home_y20gam.stage exist, renaming for firmware update
starting pid 558, tty '': '/etc/init.d/rcK'
mount: mounting /dev/mtdblock3 on /home failed: Resource busy
sh: you need to specify whom to kill`

`dispatch.c(get_config-1262)[00:00:23.136]:reset because no wifi config`

`dispatch.c(p_worker-3307)[06:41:36.097]:DISPATCH_SET_DEFAULT_TIME 1453185696
dispatch.c(p_worker-3337)[06:42:36.556]:rcd may crashed
killall: mp4_record: no process killed
dispatch.c(p_worker-3337)[06:43:37.054]:rcd may crashed
killall: mp4_record: no process killed`

So it looks like the hack applied, but no wifi config was found
Looking at the wifi config wiki it says about a yi-hack folder on the SD card,
I couldnt find it on mine anywhere so I made it and put the files in there but still didnt seem to work

Also the last 4 lines of my log just keep repeating till i stopped the camera, i left for a good 20 mins and it just stayed on loop
RCD Crash, Killall mp4

roleoroleo, 2021-03-03:
The hack doesn't allow to configure wifi.
So, you need to configure wifi with the original app.
The log shows that the hack is ok, the web site on port 8080 should be running.

pomtom44, 2021-03-03:
> The hack doesn't allow to configure wifi.
> So, you need to configure wifi with the original app.
> The log shows that the hack is ok, the web site on port 8080 should be running.

but I can change the wifi using the hack once its connected?
My wifi the cameras are on is different to the wifi my phone is on so it would be great if there was a way to set it from the config rather than using the app?

roleoroleo, 2021-03-03:
No, with the current version you can't.
You need to connect phone and cam to the same network and configure it.
But you can do this even after applying the hack. Enable cloud in configuration page and use the app to connect to the cam.

roleoroleo, 2021-03-03:
I'm working on a new version where you will be able to configure wifi but I don't know when I will finish it.

pomtom44, 2021-03-03:
> No, with the current version you can't.
> You need to connect phone and cam to the same network and configure it.
> But you can do this even after applying the hack. Enable cloud in configuration page and use the app to connect to the cam.

Do you know of a hack or version which does support this?
My cameras are not internet connected so I dont use the cloud

I have some of the older models and the hacks for those support the wifi config file, meaning i can just put the hack on SD and it connects straight away, no need for the app

roleoroleo, 2021-03-03:
I think this is the only hack available for this model.
But I don't understand the problem.
After wifi configuration you don't need the app. The configuration is permanent and you can forget it.
If the cam reboots, it reconnects to the same wifi.
You need the app only if you want to change wifi.

If you don't want to use the app at all, you can configure the wifi with this procedure:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials
Prepare an sd card and use it to configure the cam.
When the procedure is over remove the sd.

pomtom44, 2021-03-03:
> I think this is the only hack available for this model.
> But I don't understand the problem.
> After wifi configuration you don't need the app. The configuration is permanent and you can forget it.
> If the cam reboots, it reconnects to the same wifi.
> You need the app only if you want to change wifi.
> 
> If you don't want to use the app at all, you can configure the wifi with this procedure:
> https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials
> Prepare an sd card and use it to configure the cam.
> When the procedure is over remove the sd.
> But this procedure works only after the hack is done.

Its more of a convience thing
I dont want to have to connect my phone to a different wifi to setup the camera 
(i have a number of security steps to keep that CCTV wifi network locked down such as super long passwords, MAC authentication on the wifi as well as IP allocation, so it takes a while to get it setup)

Those steps you linked are the ones I was talking about before?

"Looking at the wifi config wiki it says about a yi-hack folder on the SD card,
I couldnt find it on mine anywhere so I made it and put the files in there but still didnt seem to work"

and you said it cant be done, 

> The hack doesn't allow to configure wifi.
> So, you need to configure wifi with the original app.

so im confused now?

pomtom44, 2021-03-03:
Ok so managed to get to work in the end
Reset camera, and watched logs to confirm full reset

loaded hack on and watched logs (This time i saw files being modified so knew hack worked)
Loaded app and connected
Rebooted and put wifi hack on SD then rebooted camera again

Connected fine and can access webpage fine

Having another problem now with RTSP but will open new issue for that

closed 2021-03-03 by pomtom44

roleoroleo, 2021-03-04:
Ok.
Sorry, my english is not good.
I probably wasn't clear.