roleoroleo/yi-hack-Allwinner #250: Is my BFUS cam dead? Plugin in -> Instant steady orange led

DerpyDoom, 2021-03-06:
Hey!

I hacked the Yi cam quite a while ago and stored it aside since I didn't need it. I now found a place to put it but it seems like it's bricked or dead somehow.

If I plugin the cam I instantly get a steady orange led. :(

roleoroleo, 2021-03-07:
I need more information to help you.

DerpyDoom, 2021-03-07:
What kind do you need? I'm happy to provide everything you need.

roleoroleo, 2021-03-08:
I think you don't have the serial adapter connected.

Maybe you are using a wrong power adapter or cable?
If not, try to remove the hack and reset the cam.

DerpyDoom, 2021-03-08:
I don't have any serial adapter. The thing is I am not even able to reset the cam. It's always an orange led that's lit. When I hold the reset button nothing happens.

Like I said, it was mid 2020 when I got the cam. I put it aside because I wasn't using it.

closed 2021-07-17 by DerpyDoom