roleoroleo/yi-hack-Allwinner #251: FTP Continuous Write

AJStubbsy, 2021-03-08:
Hack Version: yi-hack-Allwinner 0.2.3

Camera Model: Yi Home 1080p (BFUSY31)

Camera Version: 8.2.0.0A_202012041528

Hi, I'm experiencing a problem where the camera files are constantly uploaded and writing over existing files, when I enable FTP Upload. I'm using FTP server addon for Home Assistant and the files are being written to a /media directory on a SSD attached to the HA RPi. I've set the upload file path on the camera 'Motion Events' page and I have two directories within /media e.g. /media/yicamback. This is because there's a chance the cameras could output a file with the same name. When I review the uploaded files, I can see them constantly being written over. Any idea how I can resolve this please? Is there some problem with having sub-directories maybe?
Thanks for creating this by the way, awesome work.

roleoroleo, 2021-03-08:
If I remember correctly, the script doesn't save the status about the last file uploaded.
It  sends the last n files at every run.

AJStubbsy, 2021-03-08:
Would it send the last n files and usually stop do you think?

Catfriend1, 2021-03-16:
Yeah, maybe delete after upload makes sense for the topic starter.

AJStubbsy, 2021-03-17:
That is an option but you lose the ability to scrub through the days recordings via the app and it would be good to have a back-up of the file rather than a single version. I wonder why the script continuously runs. I've tried looking through the code (I'm no developer) but I can't see anything related to the FTP server.

Catfriend1, 2021-03-26:
This should be solved by 0.2.4

AJStubbsy, 2021-03-27:
I've tested this and the problem has been resolved for me, no more continuous writes!

Thank you very much!!

Catfriend1, 2021-03-27:
Perfect, then I'd suggest you close the ticket.

AJStubbsy, 2021-03-27:
I didn't realise that was my responsibility/power 😅! Thank you for sorting this👍

closed 2021-03-27 by AJStubbsy