roleoroleo/yi-hack-Allwinner #254: Problem trying to use the yi app with the hack

matxino, 2021-03-13:
Hello:

Thank you for your effort developing this great hack for the yi hack, allwinner version.

After installing the hack with firmware version 9.0.19.09_202010291747 in a BFUS 1080p yi home AI+ camera, can't use the official app.

The camera seem to be working fine conecting to internet because can access to the web interface of yi-hack and enable ssh, ftp, etc. Nevertheless the camera is not making the final pair. I can see in the webinterface that the serial number is 

 ��������������������

How can I return to the original state? now my webcam is not loaded in the official app and is talking in chinese insted of english. I have tried to undone the hack but can´t flash an older version of the firmware.

Thank you.
[image]

roleoroleo, 2021-03-13:
To remove the hack, follow this wiki page: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack

matxino, 2021-03-13:
I have already did it but my cam is talking in chinese and can’t connect with the official app. Before flashing the allwiner version I think I had flashed another version of the hack. Yi support gave me the one_y21ga.bin but I can’t flash anything with the sd card. I had soldered the cables to access to the ssh interface with an usb to serial dongle but can’t driver doesn’t work with windows 10

roleoroleo, 2021-03-13:
If the sd card method doesn't work to flash the cam, you need the serial interface.
If you try to make a backup, does it work?
Please, share with me one_y21.ga

matxino, 2021-03-13:
How could I flash the one_y21ga.bin using the serial interface? Could you help me with that? I could obtain the backup, could I send you the backup and check it?. How could I send you the files?

Thank you

roleoroleo, 2021-03-13:
> How could I flash the one_y21ga.bin using the serial interface?

I don't know,  I have to study the file.

> I could obtain the backup, could I send you the backup and check it?

Check yhe wiki.

> How could I send you the files?

Share it here, I will remove it asap.

matxino, 2021-03-13:
Thanks for your patience.
[one_y21ga.bin](https://mega.nz/file/cl8GSBzA#tSpJAc0ODnihvL9qoREeOFPB7_OFINBmbugOia5h3rc)
[backup.zip](https://mega.nz/file/Z1cR0IIZ#7sjgUDwK-exDby8YtRGSMK3bXlOAkCcy70QT7NuDJXI)

roleoroleo, 2021-03-14:
If backup works you can restore the cam without serial connection.

But, in theory, you should be able to restore the cam with one_y21ga.bin
Simply copying it to the sd card.
The upgrade procedure should start directly in the bootloader.
I've always heard of this file but I've never seen one :) :) :)

If it doesn't work, we need to create a special sd card to overwrite all partitions.

matxino, 2021-03-18:
Imposible, I have tried different sd cards many formating tools an differents operating systems and I cannot flash the camera, How could we do a special sd card as you mention?

Thanks in advice

roleoroleo, 2021-03-18:
We should create a script similar to the one that makes the backup.
This script must overwrite home partition using dd command.
But if something goes wrong the cam will be unusable.

matxino, 2021-03-18:
And there is no way to flash the .bin file with a command through ssh with the usb dongle?, It is quite risky deleting the home partition isn’t it?

matxino, 2021-03-19:
I´m taking the risk because not working in any way so, could you please with the script? I'm not very informed about this things. Thank you.

closed 2024-05-11 by github-actions[bot]