roleoroleo/yi-hack-Allwinner #256: Feature Request: Speak through camera

thenerdnest, 2021-03-16:
I use my 5 cameras through yi-hack-allwinner only. I don't use the phone app at all (except when I initially set them up to set the wifi). I have a firewall rule setup to block all of the cameras from being able to access the internet, but they are still accessible within the network. Anyways, I was playing around with the app when setting them up, and a feature I thought was really cool / useful, was the ability to talk to my kids through the camera. Can't do that with the hack as far as I know. I recognize it's a big ask, but man it would be cool!

roleoroleo, 2021-03-16:
I know this is a cool feature.
I already tried to use internal speaker but the linux device is busy, used by yi processes.
I don't know how to solve this problem.

tunerooster, 2021-05-14:
What is the speak service referred to in:
https://github.com/roleoroleo/yi-hack_ha_integration
```
. speak service (only available if you install the internal tts engine from here https://github.com/roleoroleo/yi-hack-utils)
```
Is it related to this issue?  Can it be installed without installing the other features in the yi_hack custom component?

roleoroleo, 2021-05-14:
Yes, it's related to this issue.
The new version contains 2 different services:
- speak.sh
- speaker.sh

The first service converts a text into an audio file using an internal tts engine.
The second service plays an audio file with the correct format.

Check the wiki.

tunerooster, 2021-05-17:
> For this, I assume I would have to lobby @alienatedsec/yi-hack-v5 *@*.***>
> to implement this on the Hi3518e platform correct?
>
> Correct
>
> Would this be a difficult thing for him to do?
>
> I am not sure yet

Ok, thanks.  It sounds like it is still a work in progress.  I will sit
tight then.  Great to hear it is in the works though.  All the best...

roleoroleo, 2021-05-17:
The work is completed.
I released it in the last version.

tunerooster, 2021-05-17:
Great!  I misunderstood.  I will hope for the best on the Outdoor camera!

Mi-Sanchez, 2021-05-22:
Can you maybe integrate speak.sh and speaker.sh
 to the web-interface?
Like the PTZ or Snapshot page :-)

roleoroleo, 2021-05-22:
You can use them as a web service.
Did you try?

Mi-Sanchez, 2021-05-22:
I tried the text to speech method with a html post-form. Unfortunately without success.

roleoroleo, 2021-05-24:
I'm using RESTED component in Firefox:
[image]

Mi-Sanchez, 2021-05-24:
The file "nanotts" is missing ;-)

tunerooster, 2021-08-18:
This option is also interesting.  Is there any way to tap into this?

Mi-Sanchez, 2021-08-18:
I have discarded it and use the speak.html from the last post instead.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/99#issuecomment-858908097

roleoroleo, 2021-08-18:
Check my recent commits and files:
https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/www/httpd/htdocs/pages/speak.html
https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/www/httpd/htdocs/js/modules/speak.js
https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/www/httpd/cgi-bin/speak.sh
https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/www/httpd/cgi-bin/speaker.sh

Mi-Sanchez, 2021-08-18:
Sorry! I have overlooked it. Thank you!

roleoroleo, 2021-10-13:
To use tts you need to install the engine: https://github.com/roleoroleo/yi-hack-utils/
But the wav must work: is the format correct?
S16LE / 8 KHz, 16 bit, mono

roobbb, 2022-08-17:
Hello.
Thank you very much for implementing this feature. I really love your work 😀

I have tested on my y20ga (v0.3.0) all 3 ways: tts via speak.sh, upload wav-file via speaker.sh and play local file via speaker_file.sh. And they work fine. The page http://CamIP:8080/index.html?page=speak was very helpful for testing too.

Only speaker_file.sh made me a little bit trouble at first, because the file 'pcmvol' was not found. The export statement inside speaker_file.sh seems to be missing the path of pcmvol: /home/yi-hack/bin/ (and differs from speak.sh/ speaker.sh).

I just copied pcmvol into the path /tmp/sd/bin which is already included in the export statement of speaker_file.sh. After that, playing local files from SD-Card worked fine too.

Could you please add "/home/yi-hack/bin/" to the export statement of speaker_file.sh?

Thank you very much 😀
Kind regards
roobbb

PS: What do you think about my suggestion renaming the speak script-files like this?:
speak.sh --> speak_tts.sh
speaker.sh --> speak_uploaded.sh
speaker_file.sh --> speak_local.sh
It would imho help seeing more easy, what each is good for.

<hr>
My testings via curl look like this <br>

tts: `curl -X POST -d "Hello there" "http://192.168.222.41:8080/cgi-bin/speak.sh?lang=en-US&voldb=2.0"`

upload: `curl -X POST "http://CamIP:8080/cgi-bin/speaker.sh?voldb=12.0" -F "file=@/local/path_of_some_audio /testmessage.wav"`

sd-card-file: `curl -X POST -d "Who_is_George_Orwell.wav" "http://192.168.222.41:8080/cgi-bin/speaker_file.sh?voldb=2.0"`

testing scripts via ssh:
`echo "test.wav" | /home/yi-hack/www/cgi-bin/speaker_file.sh`
`echo "I watch ya all" | /home/yi-hack/www/cgi-bin/speak.sh`

Error message from speaker_file.sh was on cli:
```
root@MyCam:~# echo "testme.wav" | /home/yi-hack/www/cgi-bin/speaker_file.sh
Content-type: application/json

/home/yi-hack/www/cgi-bin/speaker_file.sh: line 76: pcmvol: not found
{
"error":"false",
"description":"testme.wav"
}
```
Generating voice files was easy for me via https://freetts.com - I just downloaded them, changed them into S16LE with Audacity and stored the files on the Cam's SD-Card.

roleoroleo, 2022-08-18:
> Only speaker_file.sh made me a little bit trouble at first, because the file 'pcmvol' was not found. The export statement inside speaker_file.sh seems to be missing the path of pcmvol: /home/yi-hack/bin/ (and differs from speak.sh/ speaker.sh).

You are right, it's a copy&paste error from Allwinner-v2.
I will fix it.
Thank you for your contribution.

> PS: What do you think about my suggestion renaming the speak script-files like this?:
> speak.sh --> speak_tts.sh
> speaker.sh --> speak_uploaded.sh
> speaker_file.sh --> speak_local.sh
> It would imho help seeing more easy, what each is good for.

It's a little breaking change.
It could be a problem for other users that are using the current names.

closed 2024-05-11 by github-actions[bot]