roleoroleo/yi-hack-Allwinner #257: Call webhook on motion

ivanfor, 2021-03-18:
Hi.
What would be the best way of calling a webhook after motion? I don't want to rely on mqtt only for this.
I've been reading #30 and seems that ipc_read should be the way, shouldn't it?

roleoroleo, 2021-03-18:
ipc_read is able to read the queue (using a queue number > 0).
But it depens on what action you want to run.
Try to use the stdout of the command.

ivanfor, 2021-03-19:
There's something weird when you pipe stdout of ipc_read to whetever, it seems to get stuch after the first read.
Next examples are taken at the same time:

```
root@yi1080:/tmp/sd# ipc_read -n 5
MOTION_START
MOTION_STOP
MOTION_START
MOTION_STOP
MOTION_START
MOTION_STOP

```
```
root@yi1080:~# ipc_read -n 6 > /tmp/queue &
root@yi1080:~# tail -F /tmp/queue
MOTION_START
```
```
root@yi1080:~# ipc_read -n 7 | while read queue ; do echo $queue ; done
MOTION_START
```
I just can't understand this behavior. Can anybody try to reproduce it? Am I going crazy? ;)
Can the "-n" number have something to do? I've tried several, but based of what is stated in #30, any number between 4 and 9 can be used indistinctly.

P.S.: tried after a reboot with no changes.

roleoroleo, 2021-03-19:
ipc_multiplexer reads from the original queue /ipc_dispatch and copies the messages to /ipc_dispatch_x where x is a number 0-9
mqttv4 uses /ipc_dispatch_1
/ipc_dispatch_2 to /ipc_dispatch_9 are free.
When ipc_read starts, it clears the message queue.

ivanfor, 2021-03-20:
I've made some progress.
First, although I don't fully understand the reason why stdout doesn't work when piped, it gets fixed with a fflush(stdout); in ipc_read.c.
```
root@xcompiler:~/crosscompiler/yi1080allwinner/yi-hack-Allwinner/src/ipc_cmd/ipc_cmd# git diff ipc_read.c 
diff --git a/src/ipc_cmd/ipc_cmd/ipc_read.c b/src/ipc_cmd/ipc_cmd/ipc_read.c
index 11b0070..77aedb9 100644
--- a/src/ipc_cmd/ipc_cmd/ipc_read.c
+++ b/src/ipc_cmd/ipc_cmd/ipc_read.c
@@ -76,6 +76,7 @@ static int parse_message(char *msg, ssize_t len)
 
     msg[len] = '\0';
     printf("%s\n", msg);
+    fflush(stdout);
 
     return 0;
 }
```
@roleoroleo , I can send a PR if you want, although being a so simple change it isn't probably worth it.
Besides, in case someone is interested, I've done a script in order to get a webhook called in motion events:
```
#!/bin/bash

. /home/yi-hack/script/env.sh

QUEUE=9   
URL_MON='http://whatever.url/you/need/to/be/called'
URL_MOFF=''

ipc_read -n $QUEUE | while read event ; do
	case $event in
		"MOTION_START")
			echo "Motion start!!"
			[ -z $URL_MON ] || wget -q -O - $URL_MON
		;;
		"MOTION_STOP")
			echo "Motion end!!"
			[ -z $URL_MOFF ] || wget -q -O - $URL_MOFF
		;;
		*)
			echo "$line - Not implemented"
		;;
	esac
done
```

roleoroleo, 2021-03-20:
I will add the fflush
Thank you.

closed 2024-05-11 by github-actions[bot]