roleoroleo/yi-hack-Allwinner #261: how can i edit ftppush.sh on working camera ?

EvgenPavlyuchek, 2021-03-22:
i tried to edit file with ftp but cant save changes - ftruncate: No space left on device
can i connect with ssh, stop some services and edit sh file ? i see there error

i have problem with ftp upload records on my router with padavan and ftp server on it
camera upload records on pc's filezilla server normal, but dont want upload on router - error create
in same time i can create folders and files on router with other ftp client 

log on router
OK LOGIN: Client "192.168.1.128"
FAIL MKDIR: Client "192.168.1.128", "/FTP/koridor//2021Y03M22D18H"
OK LOGIN: Client "192.168.1.128"
FAIL UPLOAD: Client "192.168.1.128", "/FTP/koridor/2021Y03M22D18H/29M28S32.mp4"

i think router can be more error sensitive than filezilla server

i want to delete two "//"

```
		FTP_DIR="${FTP_DIR}/"
	fi
	#
	if [ "${FTP_DIR_TREE}" == "yes" ]; then
		if [ ! -z "${FTP_DIR_HOUR}" ]; then
			# Create hour directory on FTP server
			echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}/${FTP_DIR_HOUR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR_HOUR}"
			FTP_DIR_HOUR="${FTP_DIR_HOUR}/"
		fi
	fi
```

Catfriend1, 2021-03-26:
It works for me using WinScp in ftp mode.

EvgenPavlyuchek, 2021-03-27:
dont work
try to edit /home/yi-hack/script/ftppush.sh

Catfriend1, 2021-03-27:
Alternative is to copy it to sdcard and chmod +x it.

roleoroleo, 2021-03-27:
Or login with ssh and edit the file with vi.

EvgenPavlyuchek, 2021-03-30:
hmm
i tried to edit with vi
say error - you need to give 755 to folder
i tried
chmod 755 script
chmod: script: No space left on device

roleoroleo, 2021-03-30:
This is not good.
Are you sure you are in the correct folder /home/yi-hack/script ?

roleoroleo, 2021-03-30:
It is not normal for the file system to be full.
Check for any strange files or reapply the hack again.

EvgenPavlyuchek, 2021-03-30:
hmm
it's hard to get to the camera physically
I'll wait for a new firmware update, where I hope the extra slash will be removed
last time she refreshed normally by air

EvgenPavlyuchek, 2021-04-15:
roleoroleo
you didnt repair bug with double slash in last update

closed 2024-05-11 by github-actions[bot]