roleoroleo/yi-hack-Allwinner #262: MQTT login issue

Rushmed, 2021-03-22:
Since the updtae from 0.2.3 to 0.2.4 I get MQTT login error.
`
(7343) Client [856] has invalid password(undefined) or username(undefined)
`
On 0.2.3 it was working fine.
I checked login and password at the webinterface and in mqttv4.conf. Both are correct. Also checked the credentials in my broker.
Tried changing password on cam and broker without success.
Also did a reset of the hack and of course several reboots.

Got the same issue on latest [MStar](https://github.com/roleoroleo/yi-hack-MStar/issues/333).

Rushmed, 2021-03-24:
Downgraded all my cameras. Now the issue is gone. Hopefully there will be changes in the next release according to this.
I`ll also look for updates for ioBroker which I use for receiving the MQTT messages.

roleoroleo, 2021-03-24:
I'm pretty sure that the problem is mqtt-config.
If it's true, it does not break the functionality of the cam.
I will fix it in the next release.

Rushmed, 2021-03-25:
If needed I´ll be there for testing.

Rushmed, 2021-04-13:
Solved with version 0.2.5.
Thanks!

closed 2021-04-13 by Rushmed