roleoroleo/yi-hack-Allwinner #263: 0.2.4 broke Monocle gateway on all yi-hack cams here

FMudgen, 2021-03-23:
All yi-hack cams broke on yi-hack 0.2.4 update.

Just a heads up, RTSP stream working in VLC.

 Log looks the same right through the RTSP ENDPOINT SOCKET CONNECTED but then client starts firing DESCRIBE requests and getting 404ed. Monocle currently unusable with 70% of camera base down.

@mudgen
Mar 22 21:30:43 home monocle-gateway[39554]: -------------------------------------------------
Mar 22 21:30:43 home monocle-gateway[39554]: INITIALIZE RTSP STREAM: Driveway
Mar 22 21:30:43 home monocle-gateway[39554]: -------------------------------------------------
Mar 22 21:30:43 home monocle-gateway[39554]: - NAME : Driveway
Mar 22 21:30:43 home monocle-gateway[39554]: - LABEL : PRIMARY
Mar 22 21:30:43 home monocle-gateway[39554]: - URL : rtsp://192.168.0.55:554/ch0_0.h264
Mar 22 21:30:43 home monocle-gateway[39554]: - UUID : ST
REAM:c36e92b0-aced-4513-ac62-6e646fb60633
Mar 22 21:30:43 home monocle-gateway[39554]: - SESS : aabbb747-177d-49dd-a2a3-a227c6c252a1
Mar 22 21:30:43 home monocle-gateway[39554]: - MODIF : Mon Mar 22 2021 21:28:31 GMT-0400 (EDT)
Mar 22 21:30:43 home monocle-gateway[39554]: - TAGS : @proxy
Mar 22 21:30:43 home monocle-gateway[39554]: -------------------------------------------------

Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.474Z [INFO] [192.168.0.9:58714 <BJhaVa84d>] RTSP CLIENT SOCKET CONNECTED
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.603Z [INFO] [192.168.0.9:58714 <BJhaVa84d>] RTSP CLIENT ATTACHED TO STREAM: Driveway (STREAM:c36e92b0-aced-4513-ac62-6e646fb60633)
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.605Z [INFO] [192.168.0.9:58714 <BJhaVa84d>] RTSP ENDPOINT SOCKET CONNECTED {127.0.0.1:8554}
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.606Z [DEBUG] [192.168.0.9:58714 <BJhaVa84d>] [CLIENT REQUEST] --> [DESCRIBE] rtsp://da8f38cb-ccce-49d8-97be-5c723b06f3e4.mproxy.io:443/STREAM:c36e92b0-aced-4513-ac62-6e646fb60633?session=aabbb747-177d-49dd-a2a3-a227c6c252a1
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.606Z [TRACE] [192.168.0.9:58714 <BJhaVa84d>] [CLIENT REQUEST] --> [HEADERS] {
Mar 22 21:30:44 home monocle-gateway[39554]: “accept”: “application/sdp”,
Mar 22 21:30:44 home monocle-gateway[39554]: “user-agent”: “stagefright/1.2 (Linux;Android 5.1.1)”,
Mar 22 21:30:44 home monocle-gateway[39554]: “cseq”: “1”
Mar 22 21:30:44 home monocle-gateway[39554]: }
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.610Z [DEBUG] [192.168.0.9:58714 <BJhaVa84d>] [ENDPOINT REQUEST] --> [DESCRIBE] rtsp://da8f38cb-ccce-49d8-97be-5c723b06f3e4.mproxy.io:443/STREAM:c36e92b0-aced-4513-ac62-6e646fb60633
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.610Z [TRACE] [192.168.0.9:58714 <BJhaVa84d>] [ENDPOINT REQUEST] --> [HEADERS] {
Mar 22 21:30:44 home monocle-gateway[39554]: “accept”: “application/sdp”,
Mar 22 21:30:44 home monocle-gateway[39554]: “user-agent”: “stagefright/1.2 (Linux;Android 5.1.1)”,
Mar 22 21:30:44 home monocle-gateway[39554]: “cseq”: “1”
Mar 22 21:30:44 home monocle-gateway[39554]: }
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.611Z [DEBUG] [192.168.0.9:58714 <BJhaVa84d>] [ENDPOINT RESPONSE] <-- [404 (File Not Found, Or In Incorrect Format)] <cseq=1> (session=undefined)
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.612Z [TRACE] [192.168.0.9:58714 <BJhaVa84d>] [ENDPOINT RESPONSE] <-- [HEADERS] {
Mar 22 21:30:44 home monocle-gateway[39554]: “cseq”: “1”,
Mar 22 21:30:44 home monocle-gateway[39554]: “date”: “Tue, Mar 23 2021 01:30:44 GMT”
Mar 22 21:30:44 home monocle-gateway[39554]: }
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.612Z [WARN] [192.168.0.9:58714 <BJhaVa84d>] RTSP ENDPOINT RESPONSE ERROR: [DESCRIBE] = 404 (File Not Found, Or In Incorrect Format)
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.612Z [DEBUG] [192.168.0.9:58714 <BJhaVa84d>] [CLIENT RESPONSE] <-- [404 (File Not Found, Or In Incorrect Format)] <cseq=1> (session=undefined)
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.612Z [TRACE] [192.168.0.9:58714 <BJhaVa84d>] [CLIENT RESPONSE] <-- [HEADERS] {
Mar 22 21:30:44 home monocle-gateway[39554]: “cseq”: “1”,
Mar 22 21:30:44 home monocle-gateway[39554]: “date”: “Tue, Mar 23 2021 01:30:44 GMT”
Mar 22 21:30:44 home monocle-gateway[39554]: }
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.613Z [INFO] [192.168.0.9:58714 <BJhaVa84d>] RTSP ENDPOINT SOCKET CLOSED [192.168.0.9:58714 <BJhaVa84d>]
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.621Z [INFO] [192.168.0.9:58714 <BJhaVa84d>] RTSP CLIENT SOCKET CLOSED
Mar 22 21:30:44 home monocle-gateway[39554]: 2021-03-23T01:30:44.622Z [INFO] [192.168.0.9:58714 <BJhaVa84d>] RTSP CLIENT DETACHED FROM STREAM: Driveway (STREAM:c36e92b0-aced-4513-ac62-6e646fb60633)

Mar 22 21:31:02 home monocle-gateway[39554]: 2021-03-23T01:31:02.605Z [INFO] [RTSP PROXY] STREAM [Driveway/PRIMARY] WILL BE DE-REGISTERED IN 4 MINUTES

roleoroleo, 2021-03-23:
I have made a lot of changes about RTSP in the last version.
I don't know which one broke Monocle.
Restore 0.2.3 for now.

FMudgen, 2021-03-25:
Thanks for the ack.  Only way I know to do rollback is to pull the sd cards from all cams and put them back to a 0.2.3 hack and reinstall the cards.  Is there an online means?  A couple are in hard places for an old man to get to.

roleoroleo, 2021-03-25:
You can do the same operation remotely copying files with ftp.

FMudgen, 2021-03-25:
I've apparently been retired for too long to figure that out.  FTPed to a cam, went to the SD mount, blew away contents, transferred Factory and newhome folders and home_y20gam.stage in, and rebooted.  No update.  Cam did not come back online.

Pulled the card and cleared then copied in the folders and .stage file, replaced and powered up.  Rolled back to 0.2.3 as expected.

BUT Monocle is still broken, so 0.2.4 may not be the culprit.

Arkady23, 2021-03-28:
@FMudgen Have you changed the http-port?
The update is frozen. After the sd pulled out, all my settings were replaced with the default settings. Now again http is not 80, but 8080, etc. RTSP does not show. By the way, Legacy FTP didn't work before and it doesn't work now either.
After pulling out the sd, everything booted up. Next, you need to remove the installation files from the sd manually. However, the Web informs us that we have installed version 0.2.4. The desired settings also need to be reinstalled. I think that the RSTP program is currently being actively worked on, so soon dear Roleo will do everything.

FMudgen, 2021-03-28:
**@Arkady23**
I can't quite follow what you are saying, but with no configuration changes to Monocle, Alexa decided to start working about a day and a half after the rollback to 0.2.3.

Arkady23, 2021-03-28:
@FMudgen I didn't understand you at first :-)

sermayoral, 2021-03-30:
For me, 0.2.4 doesn't work. It doesn't stream RTSP. I haver returned to 0.2.3. 

I have just read the @Arkady23 message, and maybe the problem was the http port, but it doesn't matter now because it's working well in 0.2.3. 

I agree with @Arkady23, RTSP is changing right now, so we need to wait until @roleoroleo get a more stable version.

Thanks for your work, @roleoroleo :-)

FMudgen, 2021-03-30:
For me, 0.2.4 RTSP worked fine with VLC, but Alexa wouldn't show the stream.

roleoroleo, 2021-05-15:
Could you please send me a feedback about this issue with the last 0.2.6?

FMudgen, 2021-05-15:
0.2.6 yields same results on one cam.  I'll have to take the enclosure apart to get to the SD card to roll it back again.

FMudgen, 2021-05-16:
Here's the debug tail for 0.2.6.  From the point of the first "RTSP ENDPOINT PLAYING MEDIA STREAM", I can hear audio for 30 seconds or so up until "RTSP CLIENT SOCKET CLOSED".

May 16 12:49:54 home monocle-gateway[343073]: -------------------------------------------------
May 16 12:49:54 home monocle-gateway[343073]: INITIALIZE RTSP STREAM:  KitchenDoor
May 16 12:49:54 home monocle-gateway[343073]: -------------------------------------------------
May 16 12:49:54 home monocle-gateway[343073]:  - NAME  : KitchenDoor
May 16 12:49:54 home monocle-gateway[343073]:  - LABEL : PRIMARY
May 16 12:49:54 home monocle-gateway[343073]:  - URL   : rtsp://192.168.0.54:554/ch0_0.h264
May 16 12:49:54 home monocle-gateway[343073]:  - UUID  : STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f
May 16 12:49:54 home monocle-gateway[343073]:  - SESS  : 56bece3c-245c-45ae-859e-a94a11004a86
May 16 12:49:54 home monocle-gateway[343073]:  - MODIF : Fri May 07 2021 18:23:55 GMT-0400 (EDT)
May 16 12:49:54 home monocle-gateway[343073]:  - TAGS  : @proxy
May 16 12:49:54 home monocle-gateway[343073]: -------------------------------------------------
May 16 12:49:54 home monocle-gateway[343073]: 2021-05-16T16:49:54.440Z [INFO]  [RTSP PROXY] REGISTERING STREAM [KitchenDoor/PRIMARY]; (STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f)
May 16 12:49:54 home monocle-gateway[343073]: 2021-05-16T16:49:54.441Z [DEBUG] <RTSP-PROXY> [REQUEST]  --> [REGISTER] rtsp://192.168.0.54:554/ch0_0.h264
May 16 12:49:54 home monocle-gateway[343073]: 2021-05-16T16:49:54.441Z [TRACE] <RTSP-PROXY> [REQUEST]  --> [HEADERS] {
May 16 12:49:54 home monocle-gateway[343073]:   "cseq": "1",
May 16 12:49:54 home monocle-gateway[343073]:   "transport": "reuse_connection=0;preferred_delivery_protocol=udp;proxy_url_suffix=STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f"
May 16 12:49:54 home monocle-gateway[343073]: }
May 16 12:49:54 home monocle-gateway[343073]: 2021-05-16T16:49:54.442Z [DEBUG] <RTSP-PROXY> [RESPONSE] <-- [200 (OK)] <cseq=1> (session=undefined)
May 16 12:49:54 home monocle-gateway[343073]: 2021-05-16T16:49:54.442Z [TRACE] <RTSP-PROXY> [RESPONSE] <-- [HEADERS] {
May 16 12:49:54 home monocle-gateway[343073]:   "cseq": "1",
May 16 12:49:54 home monocle-gateway[343073]:   "date": "Sun, May 16 2021 16:49:54 GMT"
May 16 12:49:54 home monocle-gateway[343073]: }

May 16 12:49:55 home monocle-gateway[343073]: 2021-05-16T16:49:55.904Z [INFO]  [192.168.0.9:46067 <S1hVa6Cuu>] RTSP CLIENT SOCKET CONNECTED
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.030Z [INFO]  [192.168.0.9:46067 <S1hVa6Cuu>] RTSP CLIENT ATTACHED TO STREAM: KitchenDoor (STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f)
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.032Z [INFO]  [192.168.0.9:46067 <S1hVa6Cuu>] RTSP ENDPOINT SOCKET CONNECTED {127.0.0.1:8554}
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.033Z [DEBUG] [192.168.0.9:46067 <S1hVa6Cuu>] [CLIENT REQUEST]  --> [DESCRIBE] rtsp://da8f38cb-ccce-49d8-97be-5c723b06f3e4.mproxy.io:443/STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f?session=56bece3c-245c-45ae-859e-a94a11004a86
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.033Z [TRACE] [192.168.0.9:46067 <S1hVa6Cuu>] [CLIENT REQUEST]  --> [HEADERS] {
May 16 12:49:56 home monocle-gateway[343073]:   "accept": "application/sdp",
May 16 12:49:56 home monocle-gateway[343073]:   "user-agent": "stagefright/1.2 (Linux;Android 5.1.1)",
May 16 12:49:56 home monocle-gateway[343073]:   "cseq": "1"
May 16 12:49:56 home monocle-gateway[343073]: }
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.033Z [DEBUG] [192.168.0.9:46067 <S1hVa6Cuu>] [ENDPOINT REQUEST]  --> [DESCRIBE] rtsp://da8f38cb-ccce-49d8-97be-5c723b06f3e4.mproxy.io:443/STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.034Z [TRACE] [192.168.0.9:46067 <S1hVa6Cuu>] [ENDPOINT REQUEST]  --> [HEADERS] {
May 16 12:49:56 home monocle-gateway[343073]:   "accept": "application/sdp",
May 16 12:49:56 home monocle-gateway[343073]:   "user-agent": "stagefright/1.2 (Linux;Android 5.1.1)",
May 16 12:49:56 home monocle-gateway[343073]:   "cseq": "1"
May 16 12:49:56 home monocle-gateway[343073]: }
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.034Z [DEBUG] [192.168.0.9:46067 <S1hVa6Cuu>] [ENDPOINT RESPONSE] <-- [200 (OK)] <cseq=1> (session=undefined)
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.034Z [TRACE] [192.168.0.9:46067 <S1hVa6Cuu>] [ENDPOINT RESPONSE] <-- [HEADERS] {
May 16 12:49:56 home monocle-gateway[343073]:   "cseq": "1",
May 16 12:49:56 home monocle-gateway[343073]:   "date": "Sun, May 16 2021 16:49:56 GMT",
May 16 12:49:56 home monocle-gateway[343073]:   "content-base": "rtsp://127.0.0.1:8554/STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f/",
May 16 12:49:56 home monocle-gateway[343073]:   "content-type": "application/sdp",
May 16 12:49:56 home monocle-gateway[343073]:   "content-length": "636"
May 16 12:49:56 home monocle-gateway[343073]: }
May 16 12:49:56 home monocle-gateway[343073]: 2021-05-16T16:49:56.035Z [TRACE] [192.168.0.9:46067 <S1hVa6Cuu>] [ENDPOINT RESPONSE] <-- [BODY]
May 16 12:49:56 home monocle-gateway[343073]:  v=0
May 16 12:49:56 home monocle-gateway[343073]: o=- 1621183794442368 1 IN IP4 192.168.0.99
May 16 12:49:56 home monocle-gateway[343073]: s=LIVE555 Streaming Media v2018.04.25
May 16 12:49:56 home monocle-gateway[343073]: i=LIVE555 Streaming Media v2018.04.25
[... cut ...]
May 16 12:50:30 home monocle-gateway[343073]: 2021-05-16T16:50:30.595Z [INFO]  [192.168.0.9:46067 <S1hVa6Cuu>] RTSP CLIENT DETACHED FROM STREAM: KitchenDoor (STREAM:0e623268-6b93-45a2-a9a2-4e8f3c39364f)
May 16 12:50:30 home monocle-gateway[343073]: 2021-05-16T16:50:30.595Z [INFO]  [192.168.0.9:46067 <S1hVa6Cuu>] RTSP ENDPOINT SOCKET CLOSED [192.168.0.9:46067 <S1hVa6Cuu>]

May 16 12:51:04 home monocle-gateway[343073]: 2021-05-16T16:51:04.556Z [INFO]  [RTSP PROXY] STREAM [KitchenDoor/PRIMARY] WILL BE DE-REGISTERED IN 4 MINUTES

closed 2024-05-10 by github-actions[bot]

FMudgen, 2024-05-10:
That's fine.  No response from developer/maintainer in past 3 years.  No sense beating a dead horse.

roleoroleo, 2024-05-10:
I'm sorry for this.
But I have no solution for your problem.
I checked the RTSP server code and everything seems fine.