roleoroleo/yi-hack-Allwinner #264: Build P2P Client on Android

ricardojlrufino, 2021-03-23:
I wanted to develop my own P2P client, to have an application with time-based access control.
Can you help me, what tools can I be using ?!

roleoroleo, 2021-03-23:
This is an embedded system.
You have to use C language...

ricardojlrufino, 2021-03-23:
I managed to compile and run a basic hello.

The difficulty is to understand P2P, and how to distribute the stream.
Do you have any ideas or material that can help?

Related:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/41

roleoroleo, 2021-03-23:
Do you wanto to send audio/video stream like the yi original app?

ricardojlrufino, 2021-03-23:
yes ...

roleoroleo, 2021-03-23:
I don't know how to help you.
Maybe other users...

closed 2022-05-04 by ricardojlrufino

kikecalpe, 2022-10-30:
> I managed to compile and run a basic hello.
> 
> The difficulty is to understand P2P, and how to distribute the stream. Do you have any ideas or material that can help?
> 
> Related: #41

Check #362