roleoroleo/yi-hack-Allwinner #266: [VLC] audio out of sync

ricardojlrufino, 2021-03-23:
I verified that when accessing the stream through the VLC the audio is a few seconds ahead of the video.
[image]

ricardojlrufino, 2021-03-23:
After reboot cam back to normal...

closed 2021-03-23 by ricardojlrufino

roleoroleo, 2021-03-23:
There is an offset between video and audio that depends on the buffering of the rRTSPServer program.
Unfortunately it is not predictable, sometimes they are aligned, sometimes they are out of sync.

ricardojlrufino, 2021-03-23:
Thanks for the feedback.
Do you have a ffmpeg command to capture? I am having some problems in capturing the rtsp. Some software like https://antmedia.io/ the video gets corrupted