roleoroleo/yi-hack-Allwinner #268: QRCODE Scanner

ricardojlrufino, 2021-03-24:
Hi, this is my first test of making a qrcode scanner

I used the library:
https://github.com/dlbeer/quirc
With just a few tweaks to remove libpng usage, see:
[quirc.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/6197738/quirc.zip)

Here follows a compiled version, and a series of snapshots (HIGH and LOW res) for testing.
[my-qrcodes.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/6197649/my-qrcodes.zip)
using:
> export LD_LIBRARY_PATH = $ LD_LIBRARY_PATH: $ PWD
> ./qrtest -v snapshots /

or
> ./imggrabber -r low > /tmp/image.jpg
> ./qrtest -v /tmp/image.jpg

Now the challenge is to make the acquisition directly from the camera. Since I don't master C, I would waste an absurd amount of time trying to do it (I'm from Java) ..
Can anyone help?

roleoroleo, 2021-03-24:
What's the desired output?
The url in standard output?

ricardojlrufino, 2021-03-24:
the output can be a POST to a specific URL

roleoroleo, 2021-03-24:
I will investigate it, it's a nice project.

ricardojlrufino, 2021-03-25:
how fo fill buffer from bufferh264 in grayscale ?

```
int frame_decode(unsigned char *outbuffer, unsigned char *p, int length)
frame_decode(bufferyuv, bufferh264, hl_frame[res].sps_len + hl_frame[res].pps_len + hl_frame[res].idr_len) < 0)
```

roleoroleo, 2021-03-25:
Try this:
[qr_scanner.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6203183/qr_scanner.gz)

```
/*
 * Copyright (c) 2021 roleo.
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, version 3.
 *
 * This program is distributed in the hope that it will be useful, but
 * WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
 * General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/*
 * Read the last h264 i-frame from the buffer, convert it using libavcodec
 * and process with quirc.
 * The position of the frame is written in /tmp/iframe.idx
 */

#include <stdlib.h>
#include <stdio.h>
#include <ctype.h>
#include <unistd.h>
#include <sys/mman.h>
#include <getopt.h>

#ifdef HAVE_AV_CONFIG_H
#undef HAVE_AV_CONFIG_H
#endif

#include "libavcodec/avcodec.h"
#include "quirc.h"

#define BUF_OFFSET 300
#define BUF_SIZE 1786156

#define BUFFER_FILE "/dev/shm/fshare_frame_buf"
#define I_FILE "/tmp/iframe.idx"
#define FF_INPUT_BUFFER_PADDING_SIZE 32

#define RESOLUTION_HIGH 0
#define RESOLUTION_LOW 1

#define W_LOW 640
#define H_LOW 360
#define W_HIGH 1920
#define H_HIGH 1080

typedef struct {
    int sps_addr;
    int sps_len;
    int pps_addr;
    int pps_len;
    int idr_addr;
    int idr_len;
} frame;

int debug = 1;
unsigned char *addr;

void *cb_memcpy(void * dest, const void * src, size_t n)
{
    unsigned char *uc_src = (unsigned char *) src;
[... 237 lines cut ...]
            struct quirc_data data;

            quirc_extract(qr, i, &code);
            err = quirc_decode(&code, &data);
            if (err)
                printf("Decode failed: %s\n", quirc_strerror(err));
            else
                printf("Data: %s\n", data.payload);
        }

        // Unmap file from memory
        if (munmap(addr, BUF_SIZE) == -1) {
            fprintf(stderr, "Error munmapping file\n");
        } else {
            if (debug) fprintf(stderr, "Unmapping file %s, size %d, from %08x\n", BUFFER_FILE, BUF_SIZE, addr);
        }

        sleep(5);
    }

    quirc_destroy(qr);

    return 0;
}
```

ricardojlrufino, 2021-03-25:
Thanks for yout help
```
/qr_scanner 
Starting program
Mapping file /dev/shm/fshare_frame_buf, size 1786156, to b6b5b000
Decoding h264 frame
Starting decode
Decode frame
Writing yuv buffer
Segmentation fault (core dumped)

```
I get this error:

roleoroleo, 2021-03-25:
Ok, try again please:
[qr_scanner.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6205135/qr_scanner.gz)

ricardojlrufino, 2021-03-25:
I do not know exactly what I did, but it worked

[image]

ricardojlrufino, 2021-03-25:
full code

 i set low resolution by default...
```
/*
 * Copyright (c) 2021 roleo.
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, version 3.
 *
 * This program is distributed in the hope that it will be useful, but
 * WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
 * General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/*
 * Read the last h264 i-frame from the buffer, convert it using libavcodec
 * and process with quirc.
 * The position of the frame is written in /tmp/iframe.idx
 */

#include <stdlib.h>
#include <stdio.h>
#include <ctype.h>
#include <unistd.h>
#include <sys/mman.h>
#include <getopt.h>

#ifdef HAVE_AV_CONFIG_H
#undef HAVE_AV_CONFIG_H
#endif

#include "libavcodec/avcodec.h"
#include "quirc.h"

#define BUF_OFFSET 300
#define BUF_SIZE 1786156

#define BUFFER_FILE "/dev/shm/fshare_frame_buf"
#define I_FILE "/tmp/iframe.idx"
#define FF_INPUT_BUFFER_PADDING_SIZE 32

#define RESOLUTION_HIGH 0
#define RESOLUTION_LOW 1

#define W_LOW 640
#define H_LOW 360
#define W_HIGH 1920
#define H_HIGH 1080

typedef struct {
    int sps_addr;
    int sps_len;
    int pps_addr;
    int pps_len;
    int idr_addr;
    int idr_len;
} frame;

int debug = 1;
unsigned char *addr;

void *cb_memcpy(void * dest, const void * src, size_t n)
{
    unsigned char *uc_src = (unsigned char *) src;
[... 241 lines cut ...]

            quirc_extract(qr, i, &code);
            err = quirc_decode(&code, &data);
            if (err)
            	fprintf(stderr, "Decode failed: %s\n", quirc_strerror(err));
            else
            	fprintf(stderr, "Data: %s\n", data.payload);
        }

        // Unmap file from memory
        if (munmap(addr, BUF_SIZE) == -1) {
            fprintf(stderr, "Error munmapping file\n");
        } else {
            if (debug) fprintf(stderr, "Unmapping file %s, size %d, from %08x\n", BUFFER_FILE, BUF_SIZE, addr);
        }

        sleep(5);
    }

    quirc_destroy(qr);

    return 0;
}

```

roleoroleo, 2021-03-25:
Yes, the memset must be
`memset(outbuffer, 0x80, c->width * c->height);`
and not
`memset(outbuffer, 0x80, c->width * c->height * 3 / 2);`
because the buffer allocated is large enough only for luma part of the image.

ricardojlrufino, 2021-03-25:
Is this really necessary in the loop?
[image]

Sometimes i get this error:
Could not open file /dev/shm/fshare_frame_buf

ricardojlrufino, 2021-03-25:
Thanks, I've already done some commits to test ..

roleoroleo, 2021-05-15:
If you are interested in this topic please read this repo:
https://github.com/ricardojlrufino/yicam_qrcode

closed 2021-05-15 by roleoroleo